BehaviorTree/BehaviorTree.CPP #271: "Any::cast failed because it is empty" better handling or logging

SteveMacenski, 2021-03-31:
Hi,

I ran into this error a bunch https://github.com/BehaviorTree/BehaviorTree.CPP/blob/f54f6d83e5c06f44eae2b4d9700f5178dbdb4159/include/behaviortree_cpp_v3/utils/safe_any.hpp#L123 `Any::cast failed because it is empty` while prototyping some work on accessing blackboard variables for ROS2 action feedback.

It turns out this occurs, in my case, when an input or output port uses the blackboard variable, so its registered on the blackboard, but hasn't yet been assigned any value. I was curious if this was intended behavior or if there was another way to handle blackboard ports or calling default constructors on the objects inserted into the blackboard by the ports such that this cast error doesn't occur.

If you call `get("name", value_to_set)` on a blackboard variable that doesn't exist at all, you don't throw this which is inconsistent behavior, in my opinion.

My request is either:
- The blackboard calls default constructors on port types so that there's some default value in the port when you try to access it
- The error above is made to be more specific telling people why this has happened. It took me about a full day to figure out that was the root cause

facontidavide, 2021-04-01:
I am missing something. I tried to reproduce this issue with this test

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/tests/gtest_ports.cpp#L53-L127

What should be done differently to trigger your issue?

facontidavide, 2021-04-01:
I am not able to repsouce the issue: https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/tests/gtest_ports.cpp#L53-L127

What should I do differently to trigger the problem?

SteveMacenski, 2021-04-01:
Create a blackboard with the tree and before ticking the tree (simply so that your example nodes don't populate the ports, if they didn't populate on the first tick, then you could also tick once before and that would be OK, its nothing special about the tree not-ticking yet), try to access one of the variables put on the blackboard by the ports.

You can list all of the variables on the blackboard first and you will verify that the ports are on there, but when you access them, it throws that exception that its invalid.

This came up in Nav2 in the context of populating action feedback on the tick loop. We access the `path` variable which is a port for multiple BT nodes and it is shown on the blackboard when printing all of the keys. However if you try to access it before the planner completes and returns a `path`, the exception is thrown because key is there, but no value is

facontidavide, 2023-06-07:
Directly accessing the blackboard to initialize variables of the node is discouraged.

There is a specific tutorial that offer a better alternative: https://www.behaviortree.dev/docs/tutorial-basics/tutorial_08_additional_args

This is the intended behavior, since I do **NOT** want to populate the blackboard with default values.
I am improving the error message

commit edd67319 (2023-06-07): Fix #271: better error message
closed 2023-06-07 by facontidavide