BehaviorTree/BehaviorTree.CPP #274: Are locks in blackboard::getAny() necessary?

justin-maidbot, 2021-04-16:
Hi I'm evaluating the performance of Blackboard in BehaviorTree.CPP and was just curious why locks are necessary for reading values from the blackboard? As far as I know, you can't have a race condition if there aren't any writes?

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/include/behaviortree_cpp_v3/blackboard.h#L51-L53

facontidavide, 2021-04-25:
yes, it is necessary, because there COULD be other node writing simultaneously. A node might run in a different thread

closed 2021-04-25 by facontidavide

m0hikal, 2022-01-31:
Hi, I get a segmentation error when I write and read a simple string simultaneously to the same port.\
I already noticed the problem with larger pointer objects and wanted to test it with a string and got this error too.
Here is the error i get:
Stack trace (most recent call last) in thread 510959:
```
#11   Object "", at 0xffffffffffffffff, in 
#10   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f959bb9d292, in clone
#9    Source "/build/glibc-eX1tMB/glibc-2.31/nptl/pthread_create.c", line 477, in start_thread [0x7f959ca7e608]
#8    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f959bd5ede3, in std::error_code::default_error_condition() const
#7  | Source "/usr/include/c++/9/thread", line 195, in std::thread::_State_impl<std::thread::_Invoker<std::tuple<std::packaged_task<void ()> > > >::_M_run()
    |   194: 	void
    | > 195: 	_M_run() { _M_func(); }
    |   196:       };
    | Source "/usr/include/c++/9/thread", line 251, in std::thread::_Invoker<std::tuple<std::packaged_task<void ()> > >::operator()()
    |   249: 	  using _Indices
    |   250: 	    = typename _Build_index_tuple<tuple_size<_Tuple>::value>::__type;
    | > 251: 	  return _M_invoke(_Indices());
    |   252: 	}
    |   253:       };
    | Source "/usr/include/c++/9/thread", line 244, in void std::thread::_Invoker<std::tuple<std::packaged_task<void ()> > >::_M_invoke<0ul>(std::_Index_tuple<0ul>)
    |   242: 	  typename __result<_Tuple>::type
    |   243: 	  _M_invoke(_Index_tuple<_Ind...>)
    | > 244: 	  { return std::__invoke(std::get<_Ind>(std::move(_M_t))...); }
    |   245: 
    |   246: 	typename __result<_Tuple>::type
    | Source "/usr/include/c++/9/bits/invoke.h", line 95, in std::__invoke_result<std::packaged_task<void ()>>::type std::__invoke<std::packaged_task<void ()>>(std::packaged_task<void ()>&&)
    |    93:       using __type = typename __result::type;
    |    94:       using __tag = typename __result::__invoke_type;
    | >  95:       return std::__invoke_impl<__type>(__tag{}, std::forward<_Callable>(__fn),
    |    96: 					std::forward<_Args>(__args)...);
    |    97:     }
    | Source "/usr/include/c++/9/bits/invoke.h", line 60, in void std::__invoke_impl<void, std::packaged_task<void ()>>(std::__invoke_other, std::packaged_task<void ()>&&)
    |    58:     constexpr _Res
    |    59:     __invoke_impl(__invoke_other, _Fn&& __f, _Args&&... __args)
    | >  60:     { return std::forward<_Fn>(__f)(std::forward<_Args>(__args)...); }
    |    61: 
    |    62:   template<typename _Res, typename _MemFun, typename _Tp, typename... _Args>
      Source "/usr/include/c++/9/future", line 1551, in std::packaged_task<void ()>::operator()() [0x55c3bab6c804]
       1548:       operator()(_ArgTypes... __args)
       1549:       {
       1550: 	__future_base::_State_base::_S_check(_M_state);
      >1551: 	_M_state->_M_run(std::forward<_ArgTypes>(__args)...);
       1552:       }
       1553: 
       1554:       void
#6  | Source "/usr/include/c++/9/future", line 1423, in _M_run
    |  1421: 	    return std::__invoke(_M_impl._M_fn, std::forward<_Args>(__args)...);
    |  1422: 	};
    | >1423: 	this->_M_set_result(_S_task_setter(this->_M_result, __boundfn));
    |  1424:       }
      Source "/usr/include/c++/9/future", line 401, in std::__future_base::_State_baseV2::_M_set_result(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>, bool) [0x55c3bac0c20e]
        398: 	bool __did_set = false;
        399:         // all calls to this function are serialized,
        400:         // side-effects of invoking __res only happen once
      > 401: 	call_once(_M_once, &_State_baseV2::_M_do_set, this,
        402: 		  std::__addressof(__res), std::__addressof(__did_set));
        403: 	if (__did_set)
        404: 	  // Use release MO to synchronize with observers of the ready state.
#5  | Source "/usr/include/c++/9/mutex", line 683, in void std::call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::once_flag&, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&)
    |   681: #endif
    |   682: 
    | > 683:       int __e = __gthread_once(&__once._M_once, &__once_proxy);
    |   684: 
    |   685: #ifndef _GLIBCXX_HAVE_TLS
      Source "/usr/include/x86_64-linux-gnu/c++/9/bits/gthr-default.h", line 700, in void std::call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::once_flag&, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_ [... cut ...]  > >::value, void>::type*) const [0x55c3bac151a7]
         31: __NTH (memcpy (void *__restrict __dest, const void *__restrict __src,
         32: 	       size_t __len))
         33: {
      >  34:   return __builtin___memcpy_chk (__dest, __src, __len, __bos0 (__dest));
         35: }
         36: 
         37: __fortify_function void *
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f959bc096f7, in __nss_database_lookup
Segmentation fault (Address not mapped to object [0x6d6f646e6175])
```