BehaviorTree/BehaviorTree.CPP #276: Add tutorial for building Behavior Trees without XML

sea-bass, 2021-04-23:
I've been trying out this tool and my personal preference is to build BTs on the fly by instantiating nodes and putting them together in code. It's useful to me (and potentially others) who are trying to automatically generate BTs from action sequences output by a task and motion planner. Of course, I see the merit of the XML approach as well for more standalone cases.

Would it be worthwhile for me to work on a simple tutorial for the programmatic approach, if that's useful. If so, I can submit a PR but I wanted thoughts from the maintainers first.

facontidavide, 2021-04-25:
At some point, I should add an "advanced tutorials" session to the tutorials, to cover subjects like this.

On the other hand...
Take into account that the XML Tree creator does **a lot** of things that you might miss, like remapping ports in Subtrees. 

If I "support" the programmatic approach with a tutorial, people will start submitting bugs with "why this doesn't work when I create trees programmatically"?

sea-bass, 2021-04-25:
That makes sense.

So what would be you suggestion for building trees in a for-loop given an arbitrary sequence at runtime? Is it to dynamically create an XML string and then synthetize the tree out of that?

facontidavide, 2021-04-25:
I have the feeling that you are doing it wrong ;) 

Maybe reconsider the architecture of the tree.
Let me give you an example: 

Imagine that you have a Sequence of a robot that should visit 50 places. You **don't** create a Sequence with 50 "GoTo" children!

In this case, you would create a "PullDestinationFromQueue" that returns FAILURE if the queue is empty or SUCCESS otherwise; in the latter case, it would also write the destination in the blackboard to would be accessed a **single** "GoTo" node.

Your case might be different, but I invite you to think if there isn't a simple solution :)

If you really want to go for the programmatic tree creation, you can do it, but you have the responsibility of manually create port remapping if you use Subtrees

closed 2021-04-25 by facontidavide

sea-bass, 2021-04-25:
That is very helpful, and your suggestion would indeed work for my case -- thank you for the insights!

facontidavide, 2021-04-25:
great!