BehaviorTree/BehaviorTree.CPP #277: Create `Tree` from `TreeNode` for logging purposes

sea-bass, 2021-04-23:
If you build a behavior tree programmatically, your root node is a `TreeNode` and not a `Tree`, therefore it's not compatible with the loggers... and also because you have to call `executeTick()` rather than `tickRoot()`, so the top-level entry points being different is a little convenient.

Should there be a utility that creates a `Tree` out of a root node by recursively looking at its children? I feel this is absolutely necessary unless I'm missing a key feature in the existing code.

facontidavide, 2021-04-25:
Good point. 

Creating the tree programmatically is possible, but not a use case I encourage. On the other hand, it should be compatible with loggers, as you mentioned.

Let me think about it.

facontidavide, 2021-07-11:
Now I remember why it can't be done.

Many loggers requires the manifests: https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/include/behaviortree_cpp_v3/bt_factory.h#L136

This is the reason why I encapsulated this information into the class `Tree`.

closed 2021-07-11 by facontidavide