BehaviorTree/BehaviorTree.CPP #290: Plugin tutorial (and passing ros2 node to tree nodes)

HasanFarag, 2021-07-01:
Good Morning,

We would like to learn:

1. How to do the tree node as a plugin, because there is not tutorial about it or example.
2. How to pass ros2 node to tree nodes to create clients/publishers? It is okay to create dummy node with output port containing ros2 node ptr?
3. How to pass in runtime the data from gui (ours) to some tree nodes? It is okay to create dummy node with output port containing the gui data at the most left of the tree? 

The disadvantage that these dummy nodes are not a part of the behaviour but to pass data? Any recommedations?

qyzh195, 2021-08-02:
Hi, 
It seems I suffered from "How to use plugin correctly" as memtioned above lol.
I wanted to build all the nodes with a "register_all_node.cpp" into a shared lib .so file then linked it to a static library, which some of my nodes depend on. The only result I got was error"Could not load library: ./lib/libdummy_nodes_dyn.so: undefined symbol". 

My CMakeLists.txt was as:

```
aux_source_directory(src ACTION_SRC_CXX) 
add_library(bt_nodes_dyn SHARED  ${ACTION_SRC_CXX})
target_link_libraries(bt_nodes_dyn BT::behaviortree_cpp_v3)
target_link_libraries(bt_nodes_dyn MY_CUSTOM_LIB )
target_compile_definitions(bt_nodes_dyn PRIVATE  BT_PLUGIN_EXPORT )
```

So far I have to create nodes  mannully but still got:
```
../lib/libtaskMod.a(TaskBT.cpp.o): In function `MoveAction::ActionMove::ActionMove(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/home/qrx/work/taskWS/task/actions/include/action_move.h:31: undefined reference to `BT::CoroActionNode::CoroActionNode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, BT::NodeConfiguration const&)'
/home/qrx/work/taskWS/task/actions/include/action_move.h:31: undefined reference to `BT::CoroActionNode::~CoroActionNode()'
```

facontidavide, 2023-10-16:
There is now a specific tutorial about plugins.

For ROS2 specific code, see here: https://github.com/BehaviorTree/BehaviorTree.ROS2/tree/humble/btcpp_ros2_samples