BehaviorTree/BehaviorTree.CPP #296: Output ports in condition node

nocoinman, 2021-07-15:
Isn't condition nodes allowed to write to blackboard? Because I was trying to do so but I wasn't able to get the node to write to blackboard. No errors whatsoever both at compile time and runtime but the value corresponding to the key in blackboard seems to be empty after the `setOutput` call in the condition node.

SegfaultCreator, 2021-07-19:
Hello nocoinman

It's hard to find a solution for your problem with zero amount of code.
Can you  please provide:
1. The xml file together with the name of the ConditionNode or a screenshot of the ConditionNode in the Groot Tree
2. Your Implementation of that ConditionNode in cpp

BR 
SegfaultCreator

nocoinman, 2021-07-23:
Sorry turned out it was a mistake on my end. There were some grey areas in my understanding of how ports and blackboard keys worked.

**Erroneous tree**

```xml
<root main_tree_to_execute="BehaviorTree">
    <BehaviorTree ID="BehaviorTree">
        <Sequence>
            <SetBlackboard output_key="flag" value="off"/>
            <Condition ID="SampleCondition" flag=""/>
        </Sequence>
    </BehaviorTree>
    <TreeNodesModel>
        <Condition ID="SampleCondition">
            <output_port name="flag"/>
        </Condition>
    </TreeNodesModel>
</root>

```

**Correct tree**

```xml
<root main_tree_to_execute="BehaviorTree">
    <BehaviorTree ID="BehaviorTree">
        <Sequence>
            <SetBlackboard output_key="flag" value="off"/>
            <Condition ID="SampleCondition" flag_out="{flag}"/>
        </Sequence>
    </BehaviorTree>
    <TreeNodesModel>
        <Condition ID="SampleCondition">
            <output_port name="flag_out"/>
        </Condition>
    </TreeNodesModel>
</root>

```

I wrote the `SampleCondition` node to simply set the output to `on`. In the incorrect implementation I saved the value to the blackboard against the `flag` key and thought the node saving the `on` value to `flag` port will be reflected in the blackboard.

Anyways it works fine now, Thanks!

closed 2021-07-23 by nocoinman

SegfaultCreator, 2021-07-23:
Glad you solved your Problem.

Just for your Info: 
I dont think that in your case you need to create a _SetBlackboard_ Node before the actual condition. 

I can think of two use cases for the flag_out and both dont require the _SetBlackboard_ Node. 

Case 1: You want to set the value to _off_ if the condition fails and to _on_ if the condition succeeds. In this case you can just write:
```
SampleCondition::tick()
...
if (conditionTrue){
    setOutput("flag_out", "on");
    return BT::NodeStatus::SUCCESS;
}
else
{
    setOutput("flag_out", "off");
    return BT::NodeStatus::FAILURE;
}
```
Case 2: You want to use the blackboard key _flag_ only if your node succeeds in a successor action. Then you also don't need to create the _SetBlackboard_ node and can just write:

```
SampleCondition::tick()
...
if (conditionTrue){
    setOutput("flag_out", "on");
    return BT::NodeStatus::SUCCESS;
}
else
{
    return BT::NodeStatus::FAILURE;
}
```

BR SegfaultCreator