BehaviorTree/BehaviorTree.CPP #300: Can't find symbol [BT_RegisterNodesFromPlugin]

kevinlyu1, 2021-08-03:
Hi Davide,

I have been trying to build a collision detection node but seem to be running into issues. I am getting the error below:
`[bt_navigator-13] ERROR loading library [libsdr_behaviors__check_collision_condition.so]: can't find symbol [BT_RegisterNodesFromPlugin]
`

I have been trying to model my code after the code in the nav2_recoveries project because they already have a `isCollisionFree` function that should do this. I am just trying to pull this out so collision detection can be used as a condition node instead of being hidden within the nav2_recovery behaviors.

I have the macro to register the node in the .cpp file: 
`#include "pluginlib/class_list_macros.hpp"`
`PLUGINLIB_EXPORT_CLASS(sdr_behaviors::CheckCollisionCondition, nav2_core::Recovery)`

sdr_behaviors is the name space and CheckCollisionCondition is the class name. The .so is being created in the install folder, but for some reason I keep getting the error above. I believe this is similar to the issue #102 but that did not seem to fix my issue.

Please let me know if there is any information I can provide to help get this working!

samiamlabs, 2021-11-10:
Hi @kevinlyu1,
I have the same problem, did you find a solution?

samiamlabs, 2021-11-10:
Turns out I was missing `target_compile_definitions(${bt_plugin} PRIVATE BT_PLUGIN_EXPORT)` in my CMakeLists.txt
Not exactly sure what that does, but it seems to have fixed the problem.

closed 2023-02-03 by facontidavide