BehaviorTree/BehaviorTree.CPP #304: [Enhancement] pre_tick function in nodes and tree

simchanu29, 2021-10-05:
While developping with BehaviorTree.CPP I ran into a race condition issue when I have a lot of subscriber or action server created by my BT actions or control nodes.

For example, in a fallback with a lot of decorator childs containing a ROS subscriber (decorator that tick child if a value received in subscriber equals a parameter from port), it can trigger race condition issues when while ticking some messages are received inside some subscribers. 

The same issue can happen in a decorator or a control node implementing a actionlib server.

The need is to have a BT that can react to external triggers while keeping modular nodes. For example I would prefer not to declare subscribers and actionlib servers in the main cpp file and use blackboard to pass values to the BT. I would prefer to have actions and control that i can configure in groot to interact with ROS. 

To solve this issue, all the ROS input data (messages and actions) need to be updated before the tree is ticked. Thus the tree needs to be pre_ticked -> update all incoming data, then ticked -> act accordingly.

That's why I think a pre_tick function need to be added in control, action and decorator prototypes so if a tree.pre_tick() is called, these functions are called.

facontidavide, 2022-04-19:
There is a number of things that may be should be done differently here:

- Groot is not "ROS-aware" and will not be, for the time being
- You should not use the blackboard to pass subscribers and action servers. [There is a specific tutorial for that](https://www.behaviortree.dev/tutorial_08_additional_args/).

Therefore, what you need is actually the ability to call **spin_once** periodically.
That can be done periodically in the main while-loop, before the `tickRoot`

simchanu29, 2022-04-20:
You're right, a spinOce will do the trick since in cpp subscribers are not in their own thread.
the issue will however arise if someone want an asynchronous event to influence the BT inside a BT node using a separate thread.
 
Solved for my use case so I close the issue.

closed 2022-04-20 by simchanu29