BehaviorTree/BehaviorTree.CPP #312: how to determine the whole tree is executed completely, and get the status of the whole tree?

gangyaoUBT, 2021-11-04:
Hi, I am trying to get the status when the whole tree is complete. When I am using cur_feedback_status = m_tree.tickRoot(); it returns the status of every node. Is it possible to get the success/failure status when the whole tree is complete?
Thanks for your kind help!

asasine, 2021-11-15:
If you continue ticking in a loop, your tree should eventually terminate with a non-`RUNNING` status. That would indicate the status of the root node, which is recursively dependent on its children and so on and so forth.

```cpp
auto cur_feedback_status = m_tree.tickRoot();
while (cur_feedback_status != BT::NodeStatus::RUNNING)
{
    cur_feedback_status = m_tree.tickRoot();
    // optionally sleep this thread, otherwise it may cause the process to hang
}

if (cur_feedback_status == BT::NodeStatus::SUCCESS)
{
    std::cout << "Tree completed successfully." << std::endl;
}
else if (cur_feedback_status == BT::NodeStatus::FAILURE)
{
    std::cerr << "Tree did not complete successfully." << std::endl;
}
else
{
    std::cerr << "An unknown error occurred." << std::endl;
}
```

gangyaoUBT, 2021-11-15:
Thanks for your kind help. It's very helpful. I appreciate it. 
I found a solution that is to generate a tree-node at the very end to flag that the tree is completed?
Another question: Could I execute each node in a for-loop like
for( auto& node: m_tree.nodes ){
 node.executeTick();
}
I know there are lots of controls like fallback_node, if_then_else_node, I am not sure this for loop will work?
Thanks for any feedback.

asasine, 2021-11-15:
You could do that, but I would recommend putting a Sequence node at the root of your tree instead. This would give you the same behavior where each node is ticked iteratively and would simplify your C++ by only needing to tick the tree root.

kb0n, 2021-11-25:
@gangyaoUBT Maybe you have a misconception here. `cur_feedback_status = m_tree.tickRoot();` first executes a tick(!) and then returns the state of the whole tree (== root-node).  

If this execution lead to finished tree, the return-value is either `SUCCESS` or `FAILURE`. This is how you can check if the tree is completed and also what its final state is.  
If you however invoke `m_tree.tickRoot();` once again, then the tree will start again from the beginning.

A finished tree also sets the state of root-node to `IDLE`. You could use `m_tree.rootNode().isHalted()` to check for that.  
However, this would only tell you if your tree is running or not. It would not tell you if it finished. It could also mean "not started yet" or "aborted".

=> "finished" can actually only be fetched once after `tickRoot()`. It is not just a state-fetch but also triggers an action. So If you need to check this lateron someplace else in your code, you need to store this return-value and re-use it.  
What you can check all the time is "running/idle", but this probably won't help you

gangyaoUBT, 2021-11-29:
@kb0n, thanks for your kind reply. It really helps me a lot. 
Best,
Gang

asasine, 2022-05-21:
That's what our main loop looks like as well. We also include a ROS spin in order to ensure ROS callbacks are triggered within the BT nodes

```cpp
cur_feedback_status = BT::NodeStatus::RUNNING;
while (ros::ok() && cur_feedback_status == BT::NodeStatus::RUNNING)
{
    ros::spinOnce();
    cur_feedback_status = m_tree.tickRoot();
    switch (cur_feedback_status)
    {
        // ...
    }
}
```

facontidavide, 2022-05-21:
agree with @asasine

closed 2023-02-03 by facontidavide