BehaviorTree/BehaviorTree.CPP #318: Question: Are the claims made on SMACC about BT valid for this implementation?

Levi-Armstrong, 2021-11-16:
The [SMACC Comparison](https://smacc.dev/smacc-vs-behavior-trees/) makes claims about orthogonality and concurrency, but looking through this repos tutorials it seems those claims may not be valid for this repos implementation.

SteveMacenski, 2021-11-16:
Assuming my understand of orthogonality in the SM context is correct, we've been using orthogonality in Nav2's behavior trees using BT.CPP since day 1. We have **many** leaf clients that call independent task servers doing independent things on independent hardware (at times), which is the same thing that SMACC does and is described in their [definitions document](https://smacc.dev/orthogonals/). I can't speak to concurrancy, but I see no reason a control flow node couldn't be made to do so, but I'm not sure if BT.CPP has such a BT node example -- but I don't think there's anything preventing it.

That entire article is incredibly biased, inaccurate, and overtly aggressive.

facontidavide, 2021-11-16:
I am glad you asked. I agree with Steve's comment about the article being "biased, inaccurate, and overtly aggressive".

This does not mean that BT are better than SMACC (or vice-versa). 

I am not taking any stance on this matter, because I haven't had the time, need, or desire to learn about SMACC.

I expressed my opinion about that article here: https://discourse.ros.org/t/introducing-the-smacc-state-machine-library/14963/9?u=facontidavide

The most important point, for me, is:

### we should understand which features are "implementation dependent" and which are an intrinsic property of Behavior Trees

This is a critical distinction, because an implementation can always be improved, and this is precisely what that article fails to acknowledge.

miccol, 2021-11-17:
hi @Levi-Armstrong,

Thank you for bringing this up.
I agree with Steve and Davide about the nature of that article. 

Like Davide, I tried to start a discussion with the author of that article [here](https://github.com/reelrbtx/SMACC/issues/17) with no success.

Levi-Armstrong, 2021-11-17:
Everyone, thank you for the immediate response and information which confirmed my suspicion. I will leave it up to the maintainer to close since you may prefer to leave it open.

closed 2021-11-18 by facontidavide

facontidavide, 2021-11-18:
[image]

[image]

I think that **keeping** the strikethrough text there, instead of actually replacing it, can tell us a lot about the sincerity of those corrections and the will of the author to engage in a constructive conversation.