BehaviorTree/BehaviorTree.CPP #328: Priority Fallback and Sequence Node for Deep Learning Integration

JulianKu, 2022-01-11:
I just stumbled upon issue thread #12 while looking for a way to combine Machine Learning (Hierarchical Reinforcement Learning to be precise) with Behavior Trees.
I guess the Priority Fallback Node that @facontidavide came up with in the branch he mentions could be exactly what I am looking for. I saw that it never made it to main branch and the the `priority_fallback`-branch got deleted.
Any chance to revive it?

I'd like to use a Neural Network in this Priority Fallback Node which outputs values ([0, 1]) for each child node so that the node with highest value gets tried first; upon failure of one child, the one with next highest value could be tried out (maybe down to a certain value threshold after which the entire Priority Fallback Node returns failure).

I made a similar comment in the original thread, in case that one should be reopened instead.

Thanks in advance

facontidavide, 2023-10-16:
You can implement your own PriorityNode.

I you have any specific suggestion, you can create a PR

closed 2023-10-16 by facontidavide

Royboy1998, 2024-02-26:
Hey JulianKu, is your issue solved?I've recently come to the same question.Do you have any idea how it will work now?