BehaviorTree/BehaviorTree.CPP #330: How can 2 double ports value be equal?

linxigjs, 2022-01-20:
https://github.com/BehaviorTree/BehaviorTree.CPP/blob/d7e861f7b8295f224afb0677767f4e5d28670731/include/behaviortree_cpp_v3/decorators/blackboard_precondition.h#L73
Hi, i noticed that BlackboardPreconditionNode's port value maybe int, double, or string.
In tick(), the value from 2 input ports are compared by "==".
But how can 2 double variables be compared by "=="?
Should it be replaced by a selfdefined equal()?

facontidavide, 2022-04-05:
it should

facontidavide, 2022-05-13:
Change applied. Thanks for pointing this out:

```
if( getInput("value_A", value_A) &&
    getInput("value_B", value_B) )
{
        bool is_equal = (value_B == value_A);
        if ( std::is_same<double, T>::value || std::is_same<float, T>::value )
        {
            is_equal = std::abs(value_B - value_A) <= std::numeric_limits<float>::epsilon();
        }
        if(is_equal)
        {
            return child_node_->executeTick();
        }
}
```

closed 2022-05-13 by facontidavide