BehaviorTree/BehaviorTree.CPP #347: Concurrency segfault in TreeNode::getInput

skibak, 2022-02-18:
We experience rare crashes with following backtrace

```
(gdb) backtrace
#0  0x0000007fa6be1278 in ?? () from /lib/aarch64-linux-gnu/libc.so.6
#1  0x0000007fa7911b60 in nonstd::expected_lite::expected<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > BT::Any::convert<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::enable_if<std::is_same<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::value, void>::type*) const () from /opt/amazon/devel/lib/libbehaviortree_cpp_v3.so
#2  0x0000007fa7911e7c in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > BT::Any::cast<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >() const () from /opt/amazon/devel/lib/libbehaviortree_cpp_v3.so
#3  0x0000007fa79120c4 in nonstd::expected_lite::expected<void, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > BT::TreeNode::getInput<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const () from /opt/amazon/devel/lib/libbehaviortree_cpp_v3.so
```

I believe this happens in fragment here
https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/include/behaviortree_cpp_v3/tree_node.h#L218-L227
when background process modifies value stored in blackboard.

blackboard->getAny is threadsafe, but it returns Any*, and TreeNode::getInput can access it while it is modified by other thread.

Possible solution is to return copy of Any from getAny method instead of pointer.

facontidavide, 2022-03-02:
you are correct, I need to fix this thread safety issue.

facontidavide, 2022-03-06:
Ok, I try to fix this, adding another mutex level. I am not particularly happy with it.

A better solution would require a change in the API and ABI, and to use `std::shared_mutex`, which mean rising the requirements from C++14 to C++17.

In other words, if we want to do better, we need to wait for BehaviorTree.CPP 4.0.

closed 2022-03-06 by facontidavide

skibak, 2022-03-07:
Locally I fixed it by adding BB overload
Any Blackboard::getAnyValue(const std::string& key);
And using it in TreeNode::getInput

Do you have concerns about such approach?

skibak, 2022-03-12:
I saw the fix and not sure if it going to solve my use case.
My tree nodes are typically defined in sub tree, i.e. they reference child instance of Blackboard.
Background process has reference to top level blackboard.
So tree node and background process will use different mutexes for protection, and thus still able to run concurrently.

commit e0af52bf (2022-03-06): try to fix #347