BehaviorTree/BehaviorTree.CPP #348: How do I set the value of a subtree's port to a static value in XML?

asasine, 2022-02-23:
How do I set the value of a subtree's port to a static value in XML? Given this tree structure:

```xml
<root main_tree_to_execute="OuterTree">
  <BehaviorTree ID="OuterTree">
    <Sequence>
      <!-- <SetBlackboard output_key="true" value="true" /> -->
      <SubTree ID="InnerTree" my_inner_port="true" />
    </Sequence>
  </BehaviorTree>

  <!-- return immediately if {my_inner_port} is true, otherwise delay for 5s -->
  <BehaviorTree ID="InnerTree">
    <Fallback>
      <BlackboardCheckString value_A="{my_inner_port}" value_B="true" return_on_mismatch="FAILURE">
        <AlwaysSuccess />
      </BlackboardCheckString>

      <Delay delay_msec="5000">
        <AlwaysSuccess />
      </Delay>
    </Fallback>
  </BehaviorTree>
</root>
```

The `InnerTree` delays for 5s before returning, since there's no blackboard entry named `true` in `OuterTree`. If I uncomment the `SetBlackboard` action, `InnerTree` returns immediately, since the `InnerTree`'s port `my_inner_port` is remapped to `OuterTree`'s port `true`

How do I set `InnerTree`'s blackboard `my_inner_port` statically, without needing an intermediate blackboard entry in `OuterTree`?

As an aside, I find the syntax for port remapping confusing, and have come across some bugs in my team's codebase stemming from it. A more intuitive syntax would be `<SubTree ID="InnerTree my_inner_port="{my_outer_port}" />`, which would make it clearer that `my_outer_port` comes from the blackboard and not the string `my_outer_port`. It would potentially enable this scenario, where subtree ports are set statically.

facontidavide, 2022-04-05:
you can't... yet.
I am planning to change this in the future but It may take a while before I do

benjinne, 2022-04-06:
I see a possible solution, however it would require a variable amount of inputPorts to the subtree_node.
Perhaps an easy solution is if we add a few inputs like var1, var2, var3?
https://github.com/BehaviorTree/BehaviorTree.CPP/blob/e6f241cf31c22db8e91b8376235d1c45cc1b7b25/include/behaviortree_cpp_v3/decorators/subtree_node.h#L27
Then, the subtree_node would set them on the blackboard as needed.

facontidavide, 2022-05-13:
[SubTreeplus](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/include/behaviortree_cpp_v3/decorators/subtree_node.h#L80) has been specifically created to overcome this particular issue and others.

In BT.CPP 4.0 that will be the default one and the previous will be deprecated.

closed 2022-05-13 by facontidavide

asasine, 2022-05-13:
`SubTreePlus` is exactly what I wanted, thank you!

benjinne, 2022-05-13:
Does `SubTreePlus` support multiple static variables? I'm currently setting a couple variables on the blackboard before calling a subtree and I find it gets cluttered especially when using groot. Maybe I should open a new issue

facontidavide, 2022-05-13:
> Does SubTreePlus support multiple static variables?

@benjinne it does, but I don't give any guaranty about the Groot support.