BehaviorTree/BehaviorTree.CPP #350: RetryUntilSuccessful node confusion

j-corvo, 2022-02-27:
Hello,

I am trying to run the examples in this repo and I came across a problem. You say in a commit that you resolved the typo of the RetryUntilSuccesful node (with just one 's'), therefore it is deprecated. 
I tried to use the corrected one, RetryUntilSuccessful (with two 's'), but it does not work yet. It throws the runtime error:

```
terminate called after throwing an instance of 'BT::RuntimeError'
  what():  Error at line 7: -> Node not recognized: RetryUntilSuccessful
Aborted (core dumped)
```

I just opened this issue to warn the developers that this type is not resolved yet.

In case, I am doing something wrong, please tell me.

Thank you for your availability!

facontidavide, 2022-03-06:
Which tutorial is giving you problems? T05 works fine for me

j-corvo, 2022-03-06:
@facontidavide thanks for the reply. When I git cloned this repo into my workspace it worked, so the problem is resolved :)

closed 2022-03-06 by j-corvo