BehaviorTree/BehaviorTree.CPP #352: BT Node: Set Blackboard Variable

SteveMacenski, 2022-03-08:
Hi,

I think it would be nice if we had an general action node within the BT.CPP system that allowed us to set a value on the blackboard, e.g. `<SetBlackboard name="", value=""/>` so that we can set some state or internal global variables from the BT execution reaching certain points.

Certainly, we can implement this in Nav2 easily and we may do that if needed. But this feels like something with general appeal and would be of real value to BT.CPP users at large. What do you think?

facontidavide, 2022-03-09:
https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/include/behaviortree_cpp_v3/actions/set_blackboard_node.h

???

closed 2022-03-09 by facontidavide

SteveMacenski, 2022-03-09:
*cough* oh yes, I saw that... mhm... :wink: 

Thanks!