BehaviorTree/BehaviorTree.CPP #360: `FetchContent` only introduced in CMake >= 3.11

MitchAllain, 2022-04-10:
It seems like the `FetchContent` module being used since 3.6.1 of this package was introduced in CMake >= 3.11, however `CMakeLists.txt` cites version 3.5.1 as the minimum, and notably ubuntu bionic package repositories only provide 3.10

MitchAllain, 2022-04-10:
Not sure if reverting for older CMake versions would be satisfactory for you:
```cmake
elseif(BUILD_UNIT_TESTS)
    if(${CMAKE_VERSION} VERSION_LESS "3.11.0")
      find_package(GTest REQUIRED)
    else()
      include(FetchContent)
      FetchContent_Declare(
        googletest
        URL https://github.com/google/googletest/archive/609281088cfefc76f9d0ce82e1ff6c30cc3591e5.zip
      )
      # For Windows: Prevent overriding the parent project's compiler/linker settings
      set(gtest_force_shared_crt ON CACHE BOOL "" FORCE)
      FetchContent_MakeAvailable(googletest)
    endif()
endif()
```

facontidavide, 2022-05-13:
@MitchAllain thanks, I will implement this solution

closed 2022-05-13 by facontidavide