BehaviorTree/BehaviorTree.CPP #365: setOutput gives: received signal SIGFPE, Arithmetic exception.

jonatansw, 2022-04-26:
I get  **received signal SIGFPE, Arithmetic exception.** when trying to use setOutput from a condition node.

What I believe to be the relevant code is the following:
`
BT::NodeStatus GetPoseStamped::tick()
        {
                    //code to generate goal
                    setOutput<auip_interface::AddWPGoal>("waypoint", goal);
                    return BT::NodeStatus::SUCCESS;
      }

        BT::PortsList GetPoseStamped::providedPorts()
        {
            return {BT::OutputPort<auip_interface::AddWPGoal>("waypoint"), BT::BidirectionalPort<std::string>("space")};
        }
`

the XML for the tree has the following related to the code above:
`<Condition ID="GetPose" space="{space}" waypoint="{wp}"/>`

It was working until yesterday when I made an apt update, so something seems to have caused some issues, but I am not sure if it is on my end or somewhere else. I would greatly appreciate any suggestions on what to try to solve this.

Below is output from gdb:

`Thread 1 "core_test" received signal SIGFPE, Arithmetic exception.
0x00007ffff738ef96 in std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::Blackboard::Entry>, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::Blackboard::Entry> >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >::find(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/melodic/lib/libbehaviortree_cpp_v3.so
(gdb) where
#0  0x00007ffff738ef96 in std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::Blackboard::Entry>, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::Blackboard::Entry> >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >::find(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/melodic/lib/libbehaviortree_cpp_v3.so
#1  0x00007ffff6e5ccdf in std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, BT::Blackboard::Entry, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::Blackboard::Entry> > >::find(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /home/jonatan/auip_ws/devel/lib/libauip_behaviour_tree_lib.so
#2  0x00007ffff6e60f34 in void BT::Blackboard::set<auip_interface::AddWPGoal_<std::allocator<void> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, auip_interface::AddWPGoal_<std::allocator<void> > const&) () from /home/jonatan/auip_ws/devel/lib/libauip_behaviour_tree_lib.so
#3  0x00007ffff6e5dfad in nonstd::expected_lite::expected<void, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > BT::TreeNode::setOutput<auip_interface::AddWPGoal_<std::allocator<void> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, auip_interface::AddWPGoal_<std::allocator<void> > const&) ()
   from /home/jonatan/auip_ws/devel/lib/libauip_behaviour_tree_lib.so
#4  0x00007ffff6e863d8 in AUIP::BehaviourTree::GetPoseStamped::tick() () from /home/jonatan/auip_ws/devel/lib/libauip_behaviour_tree_lib.so
#5  0x00007ffff73abf2f in BT::TreeNode::executeTick() () from /opt/ros/melodic/lib/libbehaviortree_cpp_v3.so
#6  0x00007ffff73bd186 in BT::SequenceNode::tick() () from /opt/ros/melodic/lib/libbehaviortree_cpp_v3.so
#7  0x00007ffff73abf2f in BT::TreeNode::executeTick() () from /opt/ros/melodic/lib/libbehaviortree_cpp_v3.so
#8  0x00007ffff6e71560 in BT::Tree::tickRoot() () from /home/jonatan/auip_ws/devel/lib/libauip_behaviour_tree_lib.so
#9  0x00007ffff6e6f925 in AUIP::BehaviourTree::BehaviourTreeCore::TreeTickTimer(ros::TimerEvent const&) () from /home/jonatan/auip_ws/devel/lib/libauip_behaviour_tree_lib.so
#10 0x00007ffff6e7a1b9 in boost::_mfi::mf1<void, AUIP::BehaviourTree::BehaviourTreeCore, ros::TimerEvent const&>::operator()(AUIP::BehaviourTree::BehaviourTreeCore*, ros::TimerEvent const&) const ()
   from /home/jonatan/auip_ws/devel/lib/libauip_behaviour_tree_lib.so
#11 0x00007ffff6e7906f in void boost::_bi::list2<boost::_bi::value<AUIP::BehaviourTree::BehaviourTreeCore*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, AUIP::BehaviourTree::BehaviourTreeCore, ros::TimerEvent const&>, boost::_bi::rrlist1<ros::TimerEvent const&> >(boost::_bi::type<void>, boost::_mfi::mf1<void, AUIP::BehaviourTree::BehaviourTreeCore, ros::TimerEvent const&>&, boost::_bi::rrlist1<ros::TimerEvent const&>&, int) () from /home/jonatan/auip_ws/devel/lib/libauip_behaviour_tree_lib.so
#12 0x00007ffff6e7811b in void boost::_bi::bind_t<void, boost::_mfi::mf1<void, AUIP::BehaviourTree::BehaviourTreeCore, ros::TimerEvent const&>, boost::_bi::list2<boost::_bi::value<AUIP::BehaviourTree::BehaviourTreeCore*>, boost::arg<1> > >::operator()<ros::TimerEvent const&>(ros::TimerEvent const&) () from /home/jonatan/auip_ws/devel/lib/libauip_behaviour_tree_lib.so
#13 0x00007ffff6e77267 in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, AUIP::BehaviourTree::BehaviourTreeCore, ros::TimerEvent const&>, boost::_bi::list2<boost::_bi::value<AUIP::BehaviourTree::BehaviourTreeCore*>, boost::arg<1> > >, void, ros::TimerEvent const&>::invoke(boost::detail::function::function_buffer&, ros::TimerEvent const&) ()
   from /home/jonatan/auip_ws/devel/lib/libauip_behaviour_tree_lib.so
#14 0x00007ffff7b02b07 in ros::TimerManager<ros::Time, ros::Duration, ros::TimerEvent>::TimerQueueCallback::call() () from /opt/ros/melodic/lib/libroscpp.so
#15 0x00007ffff7b27559 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/melodic/lib/libroscpp.so
#16 0x00007ffff7b292fb in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/melodic/lib/libroscpp.so
#17 0x00007ffff7b80a39 in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*) () from /opt/ros/melodic/lib/libroscpp.so
#18 0x00007ffff7b6937b in ros::spin() () from /opt/ros/melodic/lib/libroscpp.so
#19 0x00005555555670ee in main ()
`

facontidavide, 2022-04-26:
I have no clue.

jonatansw, 2022-04-27:
Rebuilding everything solved it.

Thank you for the fast suggestion and for a wonderful library.

closed 2022-04-27 by jonatansw