BehaviorTree/BehaviorTree.CPP #366: target_link_libraries error in CMakeLists.txt (BT::behaviortree_cpp_v3)

dujeong, 2022-04-27:
I use ROS foxy.

I want to use BT library in my ROS package.

So I add `target_link_libraries(${PROJECT_NAME} BT::behaviortree_cpp_v3)`

but, compile was failed.. 

```
# Build
add_executable(bt_node ${SRC_FILES})
target_link_libraries(bt_node ${BEHAVIOR_TREE_LIBRARY}  )
target_include_directories(bt_node PUBLIC ${CMAKE_SOURCE_DIR}/include) #target scope
ament_target_dependencies(bt_node rclcpp std_msgs)
```

the build error is like this:
```
Target "bt_node" links to target "ament_index_cpp::ament_index_cpp" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
```

after changing from `BT::behaviortree_cpp_v3` to `behaviortree_cpp_v3`, compile was succeeded.

what is difference between `BT::behaviortree_cpp_v3` and `behaviortree_cpp_v3`?

asasine, 2022-05-12:
Could you post the rest of your CMakeLists.txt file?

dujeong, 2022-05-13:
below is my CMakeLists.txt
```
cmake_minimum_required(VERSION 3.5)
project(bt_task)

# Default to C99
if(NOT CMAKE_C_STANDARD)
  set(CMAKE_C_STANDARD 99)
endif()

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  # add_compile_options(-Wall -Wextra -Wpedantic)
  add_compile_options(-Wextra -Wpedantic)
endif()

set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(std_msgs REQUIRED)
find_package(BehaviorTreeV3)
set(BEHAVIOR_TREE_LIBRARY "behaviortree_cpp_v3")

file(GLOB_RECURSE SRC_FILES CONFIGURE_DEPENDS
  ${CMAKE_CURRENT_SOURCE_DIR}/src/*.cpp
)

# Build
add_executable(bt_docking_node ${SRC_FILES})
target_link_libraries(bt_docking_node ${BEHAVIOR_TREE_LIBRARY}  )
target_include_directories(bt_docking_node PUBLIC ${CMAKE_SOURCE_DIR}/include) #target scope
ament_target_dependencies(bt_docking_node rclcpp std_msgs)

# Install
install(
  TARGETS bt_docking_node
  DESTINATION lib/${PROJECT_NAME})
install(
  DIRECTORY launch
  DESTINATION share/${PROJECT_NAME})

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # uncomment the line when a copyright and license is not present in all source files
  #set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # uncomment the line when this package is not in a git repo
  #set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()
```

facontidavide, 2022-05-13:
I will try to address this today. To save me some time, would you be so kind to send me a ZIP file with an hello world project that fail compiling?

dujeong, 2022-05-17:
sorry, my answer is too late..
before sending my hello world project ZIP file, I want to ask about if it's okay to use `behaviortree_cpp_v3` without `BT::` (namespace)?

if then, no problem.
Now I am using only `behaviortree_cpp_v3` without `BT::` in CMakeLists.txt, and it's okay.

facontidavide, 2022-10-24:
If you make it work... problem solved

closed 2022-10-24 by facontidavide

mabasith, 2023-01-02:
I use ROS noetic.

I want to use BT library in my ROS package.

So I add target_link_libraries(${PROJECT_NAME}_node BT::behaviortree_cpp_v3 ${catkin_LIBRARIES})

but, compile was failed..
error is like this:
/usr/bin/ld: cannot find -lBT::behaviortree_cpp_v3
collect2: error: ld returned 1 exit status

any solution

dujeong, 2023-01-02:
can you test like below:

```
...
find_package(behaviortree_cpp_v3 REQUIRED)
...
ament_target_dependencies(${PROJECT_NAME}_node behaviortree_cpp_v3)
...
```

my final CMakeLists.txt is like this.

muddassir93, 2023-01-08:
> I use ROS noetic.
>
> I want to use BT library in my ROS package.
>
> So I add target_link_libraries(${PROJECT_NAME}_node BT::behaviortree_cpp_v3 ${catkin_LIBRARIES})
>
> but, compile was failed.. error is [...]

@dujeong `ament` is a dedicated build system for ROS2.

@mabasith  For ROS1 (noetic), I could able to build and run the executable by
 
```
...
find_package(behaviortree_cpp_v3 REQUIRED)
...
target_link_libraries(${PROJECT_NAME}_node
  ${behaviortree_cpp_v3_LIBRARIES}
  ${catkin_LIBRARIES}
)
...
```