BehaviorTree/BehaviorTree.CPP #369: How to subscribe ros2 topic in AsyncActionNode

Srma, 2022-05-03:
In AsyncActionNode, I want receive topic  data, but the callback is not work

```
#ifndef B_NODE_WORK_H
#define B_NODE_WORK_H

#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"
#include "behaviortree_cpp_v3/behavior_tree.h"

using namespace std::chrono_literals;
using std::placeholders::_1;

namespace BT_NODE{
  class BWorkNode : public BT::AsyncActionNode
  {
      public:
          BWorkNode(const std::string& name, const BT::NodeConfiguration& config) : BT::AsyncActionNode(name, config)
          {
              ros_node = rclcpp::Node::make_shared("b_node_work_clien");
              _subscription = bt_node->create_subscription<std_msgs::msg::String>("v/work/process_callback", 10, std::bind(&BWorkNode::topic_callback, this, _1));
          }

          BT::NodeStatus tick() override;
          {
               _halt_requested.store(false);
               while (!_halt_requested)
                {
                    std::this_thread::sleep_for(std::chrono::milliseconds(1000));
            
                    std::cout << "wait callback" << std::endl;
                }
            
                return BT::NodeStatus::SUCCESS;
          }

          static BT::PortsList providedPorts()
          {
            return {};
          }

          virtual void halt() override
          {
              _halt_requested.store(true);
          }

      private:
          void topic_callback(const std_msgs::msg::String::SharedPtr msg)
          {
                _halt_requested.store(true);
          }
          rclcpp::Subscription<std_msgs::msg::String>::SharedPtr _subscription;
          std::shared_ptr<rclcpp::Node> ros_node;
          std::atomic_bool _halt_requested;

  }; 
}
#endif
```

asasine, 2022-05-12:
Are you spinning the ROS node between each tick?

```cpp
rclcpp::spin_some(ros_node);
tree.tickRoot();
```

facontidavide, 2022-05-13:
@asasine anticipated my question :)

Also, if you expect multiple instances of `BWorkNode` to be created in the tree, be careful because you are creating multiple subscribers and you probably want to use a singleton/static instance or dependency injection.

There are more problems here:

       std::this_thread::sleep_for(std::chrono::milliseconds(1000));

Don't. this means that there is up to 1 second latency betwen the halt signal and the actual end of the thread. Change it to 1 ms (the extra overhead is almost non existent) or, better, use [condition variable](https://en.cppreference.com/w/cpp/thread/condition_variable).

Additionally, using a **BT::AsyncActionNode** here that spawn its own thread make no sense. This can be modelled as a **BT::StatefulAction** instead.

facontidavide, 2023-06-07:
Please check https://github.com/BehaviorTree/BehaviorTree.ROS2

closed 2023-06-07 by facontidavide