BehaviorTree/BehaviorTree.CPP #370: Listening to 3-4 topics parallely as SimpleConditions

jaswanthmv, 2022-05-11:
I am pretty new to Behavior trees. I have a situation where I have to monitor 3-4 topics parallely and implement some functionality basing on that. I tried creating a SimpleCondition for each subscriber/topic and implemented it as a Parallel tree. But I have observed that each tree node is getting executed in sequence . I have looked into ReactiveSequence but it holds good for SimpleAction nodes whereas all I have are SimpleCondition nodes. Any solution for this issue?
The XML for tree is as follows:

     <BehaviorTree ID="MainTree">
        <Parallel success_threshold="2" failure_threshold="1">
            <simple_condition name="simple_condition"/>
            <CheckBattery   name="battery_ok"/>
        </Parallel>

asasine, 2022-05-12:
A `ReactiveSequence` seems good for your use case. Since conditions are synchronous, no other asynchronous actions would be ticked between each condition, and all of the conditions would be ticked sequentially. 

```xml
 <BehaviorTree ID="MainTree">
    <ReactiveSequence>
        <SimpleCondition1 />
        <SimpleCondition2 />
        <SimpleCondition3 />
        <Action1 />
    </ReactiveSequence>
</BehaviorTree>
```

Assuming `SimpleCondition1` through `SimpleCondition3` are synchronous (always return `SUCCESS` or `FAILURE`), `Action1` will only run when all of the conditions are met.

If your condition's `tick()` method is performing work (e.g., spinning a ROS node), could that be done in an asynchronous action node or between each tick outside of the tree?

facontidavide, 2022-05-13:
" But I have observed that each tree node is getting executed in sequence".

Indeed. BehaviorTree execution is single threaded, so anything is somehow "sequential". 

The **ParallelNode** execute the children in a way that allow multiple children to be in the RUNNING state, but is not parallel in the sense of multi-threading.

The solution proposed by @asasine is the default one. If that doesn't work for you, you should specify more clearly what your desired logic is.

closed 2022-05-30 by jaswanthmv