BehaviorTree/BehaviorTree.CPP #381: Problem with node cli arguments

panwauu, 2022-05-15:
I am trying to pass the tree path as a command line argument as described [here](https://answers.ros.org/question/249061/reading-rosrun-command-line-arguments-inside-your-node/). But whenever I add the ~(tilde) to the NodeHandle the tree execution fails on an instant. Sadly I have no clue why this could happen.
Thanks for any input.

How to reproduce: 
In [test_bt.cpp](https://github.com/BehaviorTree/BehaviorTree.ROS/blob/81f49f09179b6b93969dfb5a937a146a2325ed4f/test/test_bt.cpp):
```
// Change line 184:
ros::NodeHandle nh;
// to 
ros::NodeHandle nh("~");
```

asasine, 2022-05-18:
In the implementation of `BT::RosServiceNode`, which is given the `nh` instance created on line 184 of test_bt.cpp, the node handle is used to create a ROS service client

https://github.com/BehaviorTree/BehaviorTree.ROS/blob/81f49f09179b6b93969dfb5a937a146a2325ed4f/include/behaviortree_ros/bt_service_node.h#L90

I don't think the `ros::NodeHandle::serviceClient` function is valid for private node handles (created with `ros::NodeHandle nh("~")`)

facontidavide, 2022-05-18:
Thanks, I will check this out

facontidavide, 2022-05-18:
I think that a better design of my template might use dependency injection: instead of cresting a new service client, an existing one should probably be passed as a reference/pointer.
The same is true for a potential template of ROS subscriber

closed 2023-06-07 by facontidavide