BehaviorTree/BehaviorTree.CPP #384: #373 broke existing syntax for subtree inclusion

sarcasticnature, 2022-05-19:
as mentioned by @asasine in https://github.com/BehaviorTree/BehaviorTree.CPP/pull/358#issuecomment-1126438162, PR #373 broke subtree xml file inclusion if the tree has a `main_tree_to_execute` attribute in the root element.

While this can be solved by removing the `main_tree_to_execute` attribute in the file that has your subtrees to include, this prevents that file from being run on its own.  Being able to run a subtree in isolation is quite useful when you are using BT.CPP at scale, so my view is that the change around this line
https://github.com/BehaviorTree/BehaviorTree.CPP/blob/12373ad8d85286181658ea91720538cb53082083/src/xml_parsing.cpp#L433
in #373 should be reverted so that the `main_tree_to_execute` attribute is only respected in the original file, not included ones.

asasine, 2022-05-21:
We can't upgrade to the latest version of this library until this is fixed. For what it's worth, my PR contains unit tests that would have caught this, preventing the regression from being checked in.

facontidavide, 2022-05-21:
ok, will try to fix it this weekend

facontidavide, 2022-05-21:
Fixed: https://github.com/BehaviorTree/BehaviorTree.CPP/commit/284b1b79c044246051fe55452dee0b365f116714

Thanks for reporting

closed 2022-05-21 by facontidavide