BehaviorTree/BehaviorTree.CPP #385: multiple instances of a subtree/different actionID's

pdingemans, 2022-05-20:
is there way to have multiple instances of a subtree working on different C++ nodes/objects (of the same type).?

Lets elaborate a bit on this:
Lets assume I have a centipede robot with 6 arms.
These arms all expose the same behaviour, but are tied to different parts of the hardware.
To accomodate this, there's a class arm. Each object of this type is able to control one (1) arm.
The behaviour tree will control/synchronize the movement of the 6 arms.

The behaviour of an arm will be described by a subtree, the behaviour of every arm is the same.
Only the order/timing will be different, which will be handle by a sequence.

It is possible to define a subtree that implements an arm movement and attach that to an instance of an object (via the action ID).
But that actionID is fixed. So a solution(very not preferable for obvious reasons) could be to copy the subtree for every arm and change the ActionID in the subtree.

A solution could be to make the actionID parameterizable, but I don't think this is possible/ haven't found a way for doing so.

Any thoughts on how to tackle this issue?

facontidavide, 2022-05-20:
The solution is pretty easy:

- encapsulate your arm behavior into a Subtree (as you said).
- pass the ID to the Subtree as a port has explained [here](https://www.behaviortree.dev/tutorial_06_subtree_ports/)

> A solution could be to make the actionID parameterizable, but I don't think this is possible/ haven't found a way for doing so.

Why not using a port / blackboard?

pdingemans, 2022-05-20:
thanks for the reply, but that does  not seem to do the job. Here's a snippet of an example:
```
<BehaviorTree ID="ponger">
        <Sequence>
            <Action ID="SaySomething" message="{pongseqnr}"/>
            <Action ID="Ponger1" inmessage="ping" inseqnr="{pongseqnr}" outmessage="{pongresult}"/>
         .......
```

The ponger is a subtree and I want the Action ID (now its Ponger1) to be variable. I tried passing as a parameter (same as e.g. the ping variable in this example). but the runtime doesn't allow something as :  

`<Action ID="{instancename} .........">`. 

I did the experiment with this and that fails: the runtime cant find a node with name {instancename} .

Another solution would be to pass the instancename via an inputparameter (as per your suggestion) to a C++ object, but then the object should distribute the call to an appropriate handler , but that would make the C++ code overly complex.

that would look like :     

`<Action ID="Pongers" instance="{instancename}" inmessage="ping" inseqnr="{pongseqnr}" outmessage="{pongresult}"/>
`

In this solution the instancename could be Ponger1, Ponger2 etc, and the Pongers object should distribute the call to the appropriate Ponger object (1, 2 etc..) 

any other thoughts on this on how to solve this in a more elegant way?

facontidavide, 2022-05-20:
OK there is some confusion about ID (that in Action means the actual name of the action(

What I mean is: 

```XML
<Action ID="MoveArm" hardware_id="1" .. />
```

pdingemans, 2022-05-20:
yes that is possible, but then the Ponger needs to perform actions based on the ponger_id. Which I would rather avoid if possible.
c++ pseudo-code would look something like

```
void Ponger::Tick()
{
    // read the input param
  // and dispatch it to another object
  if (id == "Ponger1") ponger1.tick(....)  
  else if (id ="Ponger2") ponger2.tick(...)
  etc...
}
```
And, while we're in an object oriented system, I had hoped I could use the different instances directly from the subtree.  

Apparently that is not easily achievable.

facontidavide, 2022-05-21:
Nope

facontidavide, 2022-05-29:
By the way, if you have instances like "ponger1", "ponge2", etc and all of them have a hardcoded hardware ID (instead of a configurable one), the problem is really in your code.

closed 2022-05-29 by facontidavide