BehaviorTree/BehaviorTree.CPP #395: Possible infinite loop in retry node

aufheben1, 2022-06-13:
Retry node with while loop implementation can cause infinite loop in specific cases;
- `repeat_count` is -1 (which means infinite)
- The child of the retry node is condition node (never returns running)
- The child returns FAILURE in first tick.
- The child can return SUCCESS only when the state of the module is changed by information gathered by other module with ros2 spin.

Possible fix can be modifying the retry node logic;
- Rather than trying another loop on child failure, return RUNNING and wait for next tick.

```diff
-    while (try_index_ < max_attempts_ || max_attempts_ == -1)
+    NodeStatus child_state = child_node_->executeTick();
+    switch (child_state)
     {
-        NodeStatus child_state = child_node_->executeTick();
-        switch (child_state)
-        {
-            case NodeStatus::SUCCESS:
-            {
-                try_index_ = 0;
-                haltChild();
-                return (NodeStatus::SUCCESS);
-            }
+        case NodeStatus::SUCCESS: {
+            try_index_ = 0;
+            haltChild();
+            return (NodeStatus::SUCCESS);
+        }
 
-            case NodeStatus::FAILURE:
+        case NodeStatus::FAILURE: {
+            try_index_++;
+            haltChild();
+            if (try_index_ >= max_attempts_ && max_attempts_ != -1)
             {
-                try_index_++;
-                haltChild();
+                try_index_ = 0;
+                return (NodeStatus::FAILURE);
             }
-            break;
+        }
 
-            case NodeStatus::RUNNING:
-            {
-                return NodeStatus::RUNNING;
-            }
+        case NodeStatus::RUNNING: {
+            return (NodeStatus::RUNNING);
+        }
 
-            default:
-            {
-                throw LogicError("A child node must never return IDLE");
-            }
+        default: {
+            throw LogicError("A child node must never return IDLE");
         }
     }
-
-    try_index_ = 0;
-    return NodeStatus::FAILURE;
 }
```

But I'm worried it might affect some nodes that runs asynchronously so I need confirmation before updating the code.
Thanks :)

facontidavide, 2022-06-20:
Let me think about this.... there might be a better way to avoid locking the tree

facontidavide, 2022-06-21:
I moved the discussion here https://discourse.behaviortree.dev/t/change-looping-in-controlnodes-and-decorators-to-make-them-interruptable/53

cmraaron, 2022-07-31:
Could let the user decide?
`<Sequence policy="OneAfterTheOtherInATightLoop">`
vs
`<Sequence policy="StepByStepReturningRunningAtEachTickWhileChildrenAreSuccess">`

Some room for improvement on those policy names
Or perhaps different node names but that could grow rather quickly

asasine, 2022-08-19:
@facontidavide have you had a chance to think about this?

Side note: The https://discourse.behaviortree.dev website isn't working for me, seems like it might be down. What do you think of using [GitHub Disussions](https://resources.github.com/devops/process/planning/discussions/) as opposed to a discourse website for the community? It might foster more engagement since I see several discussion- or question-like pop up here in BT.CPP Issues.

facontidavide, 2022-08-19:
I know exactly how to solve this from a technical point of view, but this change slightly change the behavior of the node and affect some users. Instead of fixing this node, it is safer to introduce a new one, that will become default in 4,0

facontidavide, 2022-10-24:
Fixed in version 4.0. See comments here: https://www.behaviortree.dev/migration/#preemptable-nodes-and-treetickroot

closed 2022-10-24 by facontidavide