BehaviorTree/BehaviorTree.CPP #405: Initialise input ports in code

benoit-robotics, 2022-07-29:
Hello,

I can't get my head around how to initialise the input ports of a node in the c++ code. I know how to initialise a port in the xml file.

I initialise my tree this way. For the sake of simplicity there is only one node. 
```
  BT::BehaviorTreeFactory factory;
  // node
  factory.registerNodeType<PickUpObject>("PickUpObject");

  auto tree = factory.createTreeFromFile(tree_path);
  tree.tickRoot();
```

I want to find a way to set the value of an input port of the node in the code. How can I do that ?
Thank you

asasine, 2022-07-29:
You can get the blackboard of a tree using the `rootBlackboard()` method of the `BT::Tree` class

```cpp
BT::BehaviorTreeFactory factory;
// node
factory.registerNodeType<PickUpObject>("PickUpObject");

auto tree = factory.createTreeFromFile(tree_path);

// initialize some values before ticking the tree's root
auto blackboard = tree.rootBlackboard();
blackboard->set("key", "value");

tree.tickRoot();
```

The `rootBlackboard()` method is defined here https://github.com/BehaviorTree/BehaviorTree.CPP/blob/8db5593a880c60211a8871bfe4da5b2507f726b3/include/behaviortree_cpp_v3/bt_factory.h#L215

And the `BT::Blackboard` class is defined here https://github.com/BehaviorTree/BehaviorTree.CPP/blob/8db5593a880c60211a8871bfe4da5b2507f726b3/include/behaviortree_cpp_v3/blackboard.h#L23

benoit-robotics, 2022-07-30:
Thank you, I will give it a try. The constructor of a SyncActionNode takes a NodeConfiguration object and it seems to be possible to initialise the input ports through this object. But I don't know how to call this constructor using the factory object. Maybe there is another way to initialize a tree ?

benoit-robotics, 2022-08-03:
@asasine  I tried your solution but it appears that the default value of the input port ( the one that I set in the xml ) is still the one that I get when I retrieve the port value with : 
`getInput<std::string>("key").value(); `

asasine, 2022-08-11:
If you're also setting the input port in the behavior tree using the `SetBlackboard` action node, it will overwrite the blackboard entry when it's ticked.

Something worth noting is that subtrees don't share a blackboard with the root tree. If there's a subtree in the real tree you're testing, it has its own blackboard, therefore setting a key in `tree.rootBlackboard()->set("key", "value")` will not propagate to subtree blackboards automatically. You mentioned the tree you provided in this issue is simplified, so it seemed worth mentioning this caveat.

facontidavide, 2023-05-10:
The suggested solution

```c++
auto blackboard = tree.rootBlackboard();
blackboard->set("key", "value");
```

will NOT work when you have subtrees, unless that specific entry port was remapped. 

Keep blackboard scope in mind, when initializing!

The idiomatic way to do it in the XML is with **SetBlackboard** (version 3.X) or **Script** (version 4.X)

closed 2023-05-10 by facontidavide

zflat, 2023-09-13:
You could also register a simple action and bind your initial values to declared output ports like this:
```c++
factory.registerSimpleAction("GetInitialLocation",
  [this](BT::TreeNode& self) -> BT::NodeStatus {
    self.setOutput<std::string>("location", this->getInitialLocation());
    return BT::NodeStatus::SUCCESS;
  }, {
    BT::OutputPort<std::string>("location")
});
```