BehaviorTree/BehaviorTree.CPP #406: SetBlackBoard node is not working

Zhuoqian0, 2022-08-01:
Hi,
When I was using SetBlackBoard node, it was not setting the values in the blackboard as expected. 
```
<SetBlackboard output_key="detecter_id" value="Detecter"/>
```
The tick() function of set_blackboard_node.h writes setOutput("output_key", value), should it be setOutput(key, value)? Or do I need to register when using this node?
```
    virtual BT::NodeStatus tick() override
    {
        std::string key, value;
        if ( !getInput("output_key", key) )
        {
            throw RuntimeError("missing port [output_key]");
        }
        if ( !getInput("value", value) )
        {
            throw RuntimeError("missing port [value]");
        }
        setOutput("output_key", value);
        return NodeStatus::SUCCESS;
    }
```
Best Wishes！
Thank you!

ituren, 2022-08-03:
Hi,
I think you should declare your input/output key first to use it.
You can find this declaration in include/behaviortree_cpp_v3/actions/set_blackboard_node.h
`
    static PortsList providedPorts()
    {
        return { InputPort("value", "Value represented as a string. convertFromString must be implemented."),
                 BidirectionalPort("output_key", "Name of the blackboard entry where the value should be written") };
    }
`
Cheers!

Zhuoqian0, 2022-09-07:
Hi,
Thanks for your suggestion. Adding __shared_blackboard="true" solves the problem.
```
<BehaviorTree ID="MainTree" __shared_blackboard="true">
```
Cheers!

closed 2022-09-07 by Zhuoqian0