BehaviorTree/BehaviorTree.CPP #409: FeatureRequest: read values back out of subtree blackboards

cmraaron, 2022-08-06:
if we have `<SubTreePlus ID="sometree" thing="5" />` to write the value 5 into the blackboard used by `<BehaviorTree ID="sometree" />`, would it also make sense to have some thing like `<SubTreePlus ID="sometree" thing="5" keysToCopyBackIntoParentIfSuccessful="{result_of_some_calculation}"/>`, where it would expect to find that key in the child blackboard and write it back into the parent?

asasine, 2022-08-11:
If you had a `<SetBlackboard output_key="keysToCopyBackIntoParentIfSuccessful" value="42" />` in your subtree `sometree`, it should be available in the parent blackboard in the key `result_of_some_calculation`

```xml
<BehaviorTree ID="main">
  <Sequence>
    <SubTreePlus ID="sometree" thing="5" keysToCopyBackIntoParentIfSuccessful="{result_of_some_calculation}" />

    <!-- {result_of_some_calculation} should have the value "42" -->
    <SomeAction input="{result_of_some_calculation}" />
  </Sequence>
</BehaviorTree>

<BehaviorTree ID="sometree">
  <SetBlackboard output_key="keysToCopyBackIntoParentIfSuccessful" value="42" />
</BehaviorTree>
```

facontidavide, 2023-10-16:
Issue related to SubTreePlus (deprecated). Closing

closed 2023-10-16 by facontidavide