BehaviorTree/BehaviorTree.CPP #412: ROS2 example project

laurafbec, 2022-08-09:
Hi everyone
I would like to know if there is a simple package example for ROS2 in order to start learning the BehaviorTree library.
I don't have a clear idea about how should I organized the project and what is required.
Thanks in advanace

asasine, 2022-08-11:
[BehaviorTree/BehaviorTree.ROS](https://github.com/BehaviorTree/BehaviorTree.ROS) might be a good place to start. It's written in ROS1, but the organization and basic concepts should extend to ROS2 too. It has [a ROS node](https://github.com/BehaviorTree/BehaviorTree.ROS/blob/59cd301ad48c97e06253152103ec0131b4301b58/test/test_bt.cpp#L181-L206) in the test directory that creates a BT from a string, spins a ROS node, and ticks the BT until it completes successfully (or fails). Nearly all ROS nodes that execute some action using a behavior tree is going to follow a very similar pattern: create the tree then create a loop that alternates between spinning the ROS node and ticking the BT, stop ticking the BT when it completes (`BT::NodeStatus::SUCCESS` or `FAILURE`) or the ROS node is shutting down.

The [navigation2/nav2_behavior_tree](https://github.com/ros-planning/navigation2/tree/main/nav2_behavior_tree) package is a more advanced example, written in ROS2 for the Nav2 project. It provides the [`nav2_behavior_tree::BehaviorTreeEngine`](https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/behavior_tree_engine.hpp) class for easily creating and managing behavior trees. The abstraction here makes it easier to write BT-enabled ROS nodes. If you're interested in writing your own ROS2 node that uses behavior trees to perform some action, and some of the suggestions below don't fit your bill, I would recommend using this type or creating a type that's modeled after it.

The nav2_behavior_tree package also provides the [`nav2_behavior_tree::BtActionServer`](https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server.hpp) type for creating a ROS2 action server that uses the `nav2_behavior_tree::BehaviorTreeEngine` type. It's meant to be used within a [lifecycle node](https://github.com/ros2/demos/blob/humble/lifecycle/README.rst). An example node that uses this type is [`nav2_bt_navigator::BtNavigator`](https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/include/nav2_bt_navigator/bt_navigator.hpp) in [navigation2/nav2_bt_navigator](https://github.com/ros-planning/navigation2/tree/main/nav2_bt_navigator). Note: there are a couple layers of abstraction here. `nav2_bt_navigator::BtNavigator` uses a [`nav2_bt_navigator::Navigator`](https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/include/nav2_bt_navigator/navigator.hpp) which is what creates the `nav2_behavior_tree::BtActionServer`. In your own node, you could directly create a `nav2_behavior_tree::BtActionServer` and skip the middle layer.

If you don't want to use lifecycle nodes, you could create your own `BtActionServer` that functions in a similar manner, uses `nav2_behavior_tree::BehaviorTreeEngine` internally. It would be a very similar implementation, but would execute the equivalent functions for the `configure` and `activate` transitions on startup, and the `deactivate`, `cleanup`, and `shutdown` transitions on shutdown.

As for creating and structuring your behavior trees, I've noted a few common methods, in order of simplicity:
1. Write your behavior tree in code using a string of XML.
2. Write your behavior tree in an XML file, install that file as part of your build, and load that file during execution.
3. Write several behaviors as XML files and select which file to load during launch or startup based on a launch argument or ROS parameter.
4. Provide your behavior tree as a path to an XML file in a string field of a service request or action goal and load that file during execution. This method can fail or fall back to a simpler method if the file doesn't exist.

laurafbec, 2022-08-12:
@asasine, thank you very much for you comprehensive and detailed response. I'll try to check all your advices.
Thanks again!!!

paulerikf, 2022-08-12:
Just to reiterate @laurafbec's comment, this response is such a useful resource!
I can't believe my luck that this was asked and answered yesterday, as it's exactly what I needed today. 
Thanks @asasine!

laurafbec, 2023-01-14:
Hi @asasine!
I'm trying to check the Nav2 BT nodes' status from a CPP node. I'm not able to register some nodes like ComputePathToPose.

`factory.registerNodeType<nav2_behavior_tree::ComputePathToPoseAction>("ComputePathToPose");`

Is this the correct way to check the Nav2 node's status, or is there a more straightforward one?

Thanks in advance!

zeerek-polymath, 2023-01-27:
Hey guys, we developed a sandbox example to test a simple publisher and subscriber over at Polymath Robotics. It explores the concepts you brought up :)

[ros2_behavior_tree_example](https://github.com/polymathrobotics/ros2_behavior_tree_example)

SteveMacenski, 2023-06-03:
@facontidavide I think you can close this ticket

closed 2023-06-07 by facontidavide

keenan88, 2024-08-09:
Very nice & complete demo of behavior trees here, you will have to dig thru the source code to comprehend how it works, but the code is well-maintained: https://github.com/sea-bass/turtlebot3_behavior_demos/tree/main