BehaviorTree/BehaviorTree.CPP #415: BT::Tree::sleep() is missing

Migran99, 2022-08-16:
Hello everyone!
I am doing the tutorials from [here](https://www.behaviortree.dev/tutorial_04_sequence/) but I get an error compiling some examples. My compiler says that the sleep function for BT::Tree is missing, and I cannot find it anywhere.
```
error: ‘class BT::Tree’ has no member named ‘sleep’
   88 |             tree.sleep(milliseconds(10));
      |                  ^~~~~

```
Any ideas on how to solve it? The tutorials mention that this function is perefered to the std:this_thread::sleep_for() function.

Thank you!

facontidavide, 2022-08-16:
Then, you are not using the latest version of the library

Migran99, 2022-08-16:
I am using the main branch of the repo, installing as shown [here](https://github.com/BehaviorTree/BehaviorTree.CPP#how-to-compile-plain-old-cmake)

laurafbec, 2022-09-01:
The ROS2 package does not include the last version of the library?
Thanks again

Migran99, 2022-09-02:
It doesn't seem to include the latest version. At least the one I have installed (ROS 2). I haven't had time to dive into the problem yet. But if you are are just learning how to use it you can replace the sleep with the thread one, eventhough it is not recommended.

laurafbec, 2022-09-02:
Thanks @Migran99!