BehaviorTree/BehaviorTree.CPP #416: Documentation for In/Out ports?

swarajpeppermint, 2022-08-17:
I am trying to use In/Out ports in the Behavior Tree. However I can't find any documentation on the main website. Is it documented somewhere?

asasine, 2022-08-29:
There are a couple of tutorials regarding input and output ports on the main website.

- For passing input ports and using output ports, Tutorial 2 is a good place to start: https://www.behaviortree.dev/tutorial_02_basic_ports/
- For utilizing your own custom data types with ports and custom BT nodes, Tutorial 3 should be helpful: https://www.behaviortree.dev/tutorial_03_generic_ports/
- When working with ports and subtrees, Tutorial 6 talks about port remapping: https://www.behaviortree.dev/tutorial_06_subtree_ports/

Regarding port remapping and subtrees, I recommend utilizing `SubTreePlus` for subtrees as it uses a more consistent and capable form of port remapping than the standard `SubTree`
https://github.com/BehaviorTree/BehaviorTree.CPP/blob/ef68cf1c2e9f30ad63e7e5f5fe4b10fe215db6a2/include/behaviortree_cpp_v3/decorators/subtree_node.h#L40-L79

closed 2022-11-19 by facontidavide

jakobhyke, 2025-01-27:
I came across this Issue when I was looking for documentation on a combined In/Out-port.

Just in case others have the same question, you can find `BT::BidirectionalPort` [here](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/48f6c5bd1d797c73a562f1c92f404d0bb944e018/include/behaviortree_cpp/basic_types.h#L496).