BehaviorTree/BehaviorTree.CPP #417: Release cadence

asasine, 2022-08-18:
I'm curious what your anticipated release cadence is for this package. We've resorted to cloning this package as a submodule alongside our custom packages in order to receive changes soon after they're merged. An automated deployment process (e.g., daily if there are changes to the repo) to current supported ROS distros would be an amazing addition to this project.

asasine, 2022-10-25:
[REP 149](https://www.ros.org/reps/rep-0149.html#build-depend-multiple) provides attributes for the dependency tags that allow for specifying version information. What do you think of releasing BehaviorTree.CPP under a unified package name `behaviortree_cpp` with appropriate versioning information? You could release both 3.x and 4.x versions simultaneously (for a period of time, to ease migration until 3.x is fully EOL).

Users could create dependencies in their package.xml files like `<build_depend version_gte="4.0.0">behaviortree_cpp</build_depend>` to get 4.x of the library, or `<build_depend version_gte="3.0.0" version_lt="4.0.0">behaviotree_cpp</build_depend>` to get 3.x of the library. Since the library is already released under `behaviotree_cpp_v3`, using the default `<build_depend>behaviortree_cpp_v3</build_depend>` would be backwards-compatible, resulting in a 3.x version of the library.

Users may continue to add the source to their workspace through tools like git submodules, conan, vcstool, and friends if they so desire. If the release cadence into upstream repositories is timely enough, I suspect most [ROS] users would benefit greatly from the simplicity of package.xml dependencies with version attributes. A quick guide in the README for using this would be helpful here too.

facontidavide, 2022-11-19:
That is just the tip of the iceberg.

**The problem is the name of the package / debian.**

If people compile from source, `<build_depend>` tag will help them, but what will happen when they run this?

```
apt install ros-****-behavior-tree
```
I don't see an easy solution, but to create a package called **behavior-tree-v4**

asasine, 2022-12-12:
Users who run `apt install ros-<ros_version>-behavior-tree` will get the latest version for the `<ros_version>` they selected, similar to how Python, CMake, etc. are pinned to a version for a given OS release. They may still specify the version of the `behavior-tree` package if they so choose, with standard apt/deb versioning features.

That being said, if a user is installing via `ros-<ros_version>-behavior-tree`, why wouldn't they be able to install via package.xml files with `<build_depend>` tags and rosdep?

lukicdarkoo, 2023-06-06:
@facontidavide Any news on this? Maybe someone from Nav2 should join the discussion as they heavily rely on BT

facontidavide, 2023-06-06:
@lukicdarkoo , version 3.8 and 4.x need to coexist as separate packages in ROS, mostly BECAUSE of Nav2

lukicdarkoo, 2023-06-06:
Then it is settled, we are looking forward to `ros-<ros_version>-behavior-tree-v4` :)

facontidavide, 2023-06-07:
[image]

closed 2023-06-07 by facontidavide

AlexGisi, 2023-06-08:
Is there any timeline for idea when this will become available via `apt install`?

facontidavide, 2023-06-09:
That is out my control

Timple, 2023-06-09:
@AlexGisi You can monitor this categorie for this: https://discourse.ros.org/c/release/16