BehaviorTree/BehaviorTree.CPP #425: FileLogger Buffer Copy Unaligned Memory Segmentation Fault

jaelrod, 2022-09-13:
- Scenario
  The use case at hand is to iteratively send control commands to a robot (via ROS messages) until some criteria is satisfied (as communicated on a ROS topic), resulting in a behavior that inches forward very slowly until a threshold is determined to have been reached. This has been implemented in our system as the following BehaviorTree (partially redacted/modified for brevity & clarity)
```
  <BehaviorTree ID="InchForward">
    <Sequence>
      <ReactiveFallback>
        <Fallback>
          <IsThresholdReached /> <!-- custom BT node checking a simple ROS topic -->
        </Fallback>

        <ReactiveSequence>
          <Timeout msec="90000">
            <KeepRunningUntilFailure>
              <Sequence>
                <Delay delay_msec="3500">
                  <AlwaysSuccess />
                </Delay>

                <ForceSuccess>
                  <Timeout msec="500">
                    <KeepRunningUntilFailure>
                      <InchForward /> <!-- custom BT node sending a simple ROS message to inch forward -->
                    </KeepRunningUntilFailure>
                  </Timeout>
                </ForceSuccess>

                <ForceSuccess>
                  <Timeout msec="1000">
                    <KeepRunningUntilFailure>
                      <InchForward /> <!-- custom BT node inching even more slowly, this time -->
                    </KeepRunningUntilFailure>
                  </Timeout>
                </ForceSuccess>
              </Sequence>
            </KeepRunningUntilFailure>
          </Timeout>
        </ReactiveSequence>
      </ReactiveFallback>
    </Sequence>
  </BehaviorTree>
```

- Issue
  Rarely, during this inching forward behavior, one of these shorter duration `Timeout` nodes experiences its timeout, broadcasts its transitioning state into Idle status, a FileLogger who is subscribed executes its callback to put the `BT::SerializedTransition` into its buffer (buffer size in use is default 10), and while trying to `push_back` into its internal buffer, experiences a segmentation fault in a copy initialization of the `BT::SerialiezedTransition`, because of unaligned memory.

  This has been very elusive and difficult to reproduce and catch with a debugger. The most reliable way has been to engage the behavior and let all of these Timeouts timeout, but the fault still probably only occurs about 5% or less of the time. Also, it appears to happen more (maybe only?) on the first time that behavior tree is executing, although looking at the code it is unclear to me for what logical reason that would be.

  Not having looked much into BT source code until now, I am unsure exactly why this is happening, however I am suspicious it has something to do with the very short duration of these Timeout nodes (though I did not take the time to experiment with varying durations). Very short durations filling that buffer more quickly might need some thread coordination in `FileLogger` (and maybe others that use internal buffers) with `std::mutex`/etc. Seems like the issue can worked around by temporarily opting out of the buffering behavior with buffer size zero.

- Screenshot catching segmentation fault with gdb in VS Code
  [image]

  [image]

- Other details
  OS: Ubuntu 18.05.5 in WSL2, Linux - 5.10.102.1-microsoft-standard-WSL2 - x86_64
  CMake version: 3.23.2
  Compiler: gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
  BehaviorTree version: built from source, commit SHA 8db5593a880c60211a8871bfe4da5b2507f726b3
  Sample compiler invocation form build toolchain: `/usr/bin/c++ -DBT_BOOST_COROUTINE2 -DNCURSES_FOUND -DTINYXML2_DEBUG -DUSING_ROS2 -DZMQ_FOUND -Dbehaviortree_cpp_v3_EXPORTS -I/home/jaelrod/code/repo/src/robot/bridge/BehaviorTree.CPP/3rdparty -I/home/jaelrod/code/repo/src/robot/bridge/BehaviorTree.CPP/include -I/opt/ros/eloquent/include -g -fPIC   -Wpedantic -Wall -Wextra -Werror=return-type -std=gnu++14 -MD -MT CMakeFiles/behaviortree_cpp_v3.dir/src/loggers/bt_file_logger.cpp.o -MF CMakeFiles/behaviortree_cpp_v3.dir/src/loggers/bt_file_logger.cpp.o.d -o CMakeFiles/behaviortree_cpp_v3.dir/src/loggers/bt_file_logger.cpp.o -c /home/jaelrod/code/repo/src/robot/bridge/BehaviorTree.CPP/src/loggers/bt_file_logger.cpp
`

- Call stack:
```
libc.so.6!__memmove_avx_unaligned_erms() (\build\glibc-CVJwZb\glibc-2.27\sysdeps\x86_64\multiarch\memmove-vec-unaligned-erms.S:534)
libbehaviortree_cpp_v3.so!std::__copy_move<true, true, std::random_access_iterator_tag>::__copy_m<std::array<unsigned char, 12ul> >(const std::array<unsigned char, 12> * __first, const std::array<unsigned char, 12> * __last, std::array<unsigned char, 12> * __result) (\usr\include\c++\7\bits\stl_algobase.h:368)
libbehaviortree_cpp_v3.so!std::__copy_move_a<true, std::array<unsigned char, 12ul>*, std::array<unsigned char, 12ul>*>(std::array<unsigned char, 12> * __first, std::array<unsigned char, 12> * __last, std::array<unsigned char, 12> * __result) (\usr\include\c++\7\bits\stl_algobase.h:386)
libbehaviortree_cpp_v3.so!std::__copy_move_a2<true, std::array<unsigned char, 12ul>*, std::array<unsigned char, 12ul>*>(std::array<unsigned char, 12> * __first, std::array<unsigned char, 12> * __last, std::array<unsigned char, 12> * __result) (\usr\include\c++\7\bits\stl_algobase.h:424)
libbehaviortree_cpp_v3.so!std::copy<std::move_iterator<std::array<unsigned char, 12ul>*>, std::array<unsigned char, 12ul>*>(std::move_iterator<std::array<unsigned char, 12>*> __first, std::move_iterator<std::array<unsigned char, 12>*> __last, std::array<unsigned char, 12> * __result) (\usr\include\c++\7\bits\stl_algobase.h:456)
libbehaviortree_cpp_v3.so!std::__uninitialized_copy<true>::__uninit_copy<std::move_iterator<std::array<unsigned char, 12ul>*>, std::array<unsigned char, 12ul>*>(std::move_iterator<std::array<unsigned char, 12>*> __first, std::move_iterator<std::array<unsigned char, 12>*> __last, std::array<unsigned char, 12> * __result) (\usr\include\c++\7\bits\stl_uninitialized.h:101)
[... 3 lines cut ...]
libbehaviortree_cpp_v3.so!std::vector<std::array<unsigned char, 12ul>, std::allocator<std::array<unsigned char, 12ul> > >::_M_realloc_insert<std::array<unsigned char, 12ul> const&>(std::vector<std::array<unsigned char, 12>, std::allocator<std::array<unsigned char, 12> > > * const this, std::vector<std::array<unsigned char, 12>, std::allocator<std::array<unsigned char, 12> > >::iterator __position) (\usr\include\c++\7\bits\vector.tcc:431)
libbehaviortree_cpp_v3.so!std::vector<std::array<unsigned char, 12ul>, std::allocator<std::array<unsigned char, 12ul> > >::push_back(std::vector<std::array<unsigned char, 12>, std::allocator<std::array<unsigned char, 12> > > * const this, const std::vector<std::array<unsigned char, 12>, std::allocator<std::array<unsigned char, 12> > >::value_type & __x) (\usr\include\c++\7\bits\stl_vector.h:948)
libbehaviortree_cpp_v3.so!BT::FileLogger::callback(BT::FileLogger * const this, BT::Duration timestamp, const BT::TreeNode & node, BT::NodeStatus prev_status, BT::NodeStatus status) (\home\jaelrod\code\repo\src\robot\bridge\BehaviorTree.CPP\src\loggers\bt_file_logger.cpp:49)
BT::StatusChangeLogger::StatusChangeLogger(BT::TreeNode*)::{lambda(std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, BT::TreeNode const&, BT::NodeStatus, BT::NodeStatus)#1}::operator()(std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, BT::TreeNode const&, BT::NodeStatus, BT::NodeStatus) const(const BT::StatusChangeLogger::<lambda(BT::TimePoint, const BT::TreeNode&, BT::NodeStatus, BT::NodeStatus)> * const __closure, BT::TimePoint timestamp, const BT::TreeNode & node, BT::NodeStatus prev, BT::NodeStatus status) (\home\jaelrod\code\repo\install\bridge\behaviortree_cpp_v3\include\behaviortree_cpp_v3\loggers\abstract_logger.h:75)
std::_Function_handler<void (std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, BT::TreeNode const&, BT::NodeStatus, BT::NodeStatus), BT::StatusChangeLogger::StatusChangeLogger(BT::TreeNode*)::{lambda(std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, BT::TreeNode const&, BT::NodeStatus, BT::NodeStatus)#1}>::_M_invoke(std::_Any_data const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >&&, BT::TreeNode const&, [... cut ...] <BT::TimerQueue<_Clock, _Duration>::TimerQueue() [with _Clock = std::chrono::_V2::steady_clock; _Duration = std::chrono::duration<long int, std::ratio<1, 1000000000> >]::<lambda()> > > > * const this) (\usr\include\c++\7\thread:186)
libstdc++.so.6![Unknown/Just-In-Time compiled code] (Unknown Source:0)
libpthread.so.0!start_thread(void * arg) (\build\glibc-CVJwZb\glibc-2.27\nptl\pthread_create.c:463)
libc.so.6!clone() (\build\glibc-CVJwZb\glibc-2.27\sysdeps\unix\sysv\linux\x86_64\clone.S:95)
```

jaelrod, 2022-09-15:
Following up:
- Opting out of buffering with a buffer size of zero when using `FileLogger` does mitigate the crash in both AMD64 and ARM64.
- Still have not experimented with either longer or radically short timeout durations, but did confirm that the inner timeout duration of the behavior in question was updated from a single `Timeout` of 3.5 seconds to the two `Timeout` of 0.5 seconds and 1.0 seconds to increase iteration frequency and granularity. Said update was merged the very day before this segmentation fault was first observed, somewhat corroborating the theory about fast-filling buffers in a concurrent execution environment.

facontidavide, 2023-06-07:
I am not sure how to reproduce this.
If you can share an example / uni test that I can run on my computer, I will be happy to address this.

Said that, there is a new implementation: https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/include/behaviortree_cpp/loggers/bt_file_logger_v2.h

closed 2023-06-07 by facontidavide