BehaviorTree/BehaviorTree.CPP #428: Using plugins in real time to modify behavior tree.

dt1729, 2022-09-15:
Hi team,
I am new to using Behavior trees with ROS. I am trying to make a dynamic behaviour tree that changes at run time. When I say changes I mean: 

- Adding action nodes not present in the XML file but will be registered at runtime. An external code wants to trigger this change.

I have been reading about plugins and how those can be used to register nodes during runtime. Are these nodes required to be in the XML file? If yes, then where exactly(which subtree/node)? If not then where can I read more about plugins?

All advice and comments are welcome. 
Thanks 
Divya Tiwari

facontidavide, 2022-09-15:
My advice is: do not do it.

Do you modify your python script at runtime? Do you delete or add lines of code while expecting?

If you think that is nonsense, the same is true for behavior trees.

It is probably a symptom of the fact that you are not using subtrees effectively.

facontidavide, 2022-09-15:
To be more clear, I will never ever support this use case. If you manage to make this work... Go ahead! 😊

dt1729, 2022-09-15:
I was envisioning the same problem and did not find a very nice solution thus the question. I believe another way to go about this is using a Decorator with a variable conditional value governed by another function. Having this at every tree node perhaps to check this condition again and again. I believe decorators are meant for this specific use case.

asasine, 2022-12-12:
We have a solution similar to your request implemented in our ROS-enabled BT node. Our top-level node is an action server which has a client for some `my_msgs/action/Inner` action server. The node's action goal contains a `string action` field for the inner action server to call, and we have multiple action servers implementing the same ROS action interface `my_msgs/action/Inner` (e.g., `/inner/foo`, `/inner/bar`, etc.). The top-level node uses the `string action` field to create an action server to the specific inner action (e.g., `/inner/foo`), which allows us to spin up other action servers at runtime without having to change the top-level node.

Our design is similar to how Nav2 implements their [navigation servers](https://navigation.ros.org/concepts/index.html#navigation-servers). Rather than changing the tree at runtime, we change _what the tree calls_. So long as the inner action servers implement the spec correctly for their given interface `my_msgs/action/Inner`, we have confidence that the top-level tree will behave correctly, regardless of the exact inner behavior that executes.

facontidavide, 2023-06-21:
Sorry, I am not maintaining that combination.

As a developer, I will not maintain version 3.x more than STRICTLY necessary, until sponsored by a company.

I am about to release 3.9, but only because there was an actual bug that needed to be solved.

closed 2023-06-21 by facontidavide

dt1729, 2023-06-21:
I have converged to a different solution based on my needs, I have a separate monitor process that modified after a run of the BT. As my main focus was fault isolation. However, your solution is very intruiging, @asasine can you share the link to your github repository?