BehaviorTree/BehaviorTree.CPP #430: About the interface createTreeFromFile

LuossD, 2022-09-19:
**terminate called after throwing an instance of 'BT::RuntimeError'
  what():  Error parsing the XML: XML_ERROR_FILE_NOT_FOUND**

This may be a simple problem, but I can't find a solution and hope to get a reply.Thanks!

thehummingbird, 2022-09-20:
This means that the XML file wasn't present in the path you provided.
For instance, if this is how you use your XML file `factory.createTreeFromFile("./bt_tree.xml")`
Then the file wasn't in the current directory (assuming you ran from the some directory)

Please bear in mind that the path above should be relative to the directory you run from.

LuossD, 2022-09-23:
> This means that the XML file wasn't present in the path you provided. For instance, if this is how you use your XML file `factory.createTreeFromFile("./bt_tree.xml")` Then the file wasn't in the current directory (assuming you ran from the same directory)
> 
> Please bear in mind that the path above should be relative to the directory you run from.

My .cpp file and XML file are in the same directory, I call the factory. CreateTreeFromFile (". / bt_tree. XML ")  in the .cpp file,  reported this error, so the relative path to invoke the interface of CPP file path or something else? I'm so confused, thanks for your reply.

thehummingbird, 2022-09-23:
It doesn't depend on the location of your cpp file but the current working directory when you're running your executable. 
Let's say you have your executable ready after running the makefile and you run it, your CURRENT WORKING DIRECTORY (when you use your executable) should have bt_tree.xml, in your case.

LuossD, 2022-09-29:
> It doesn't depend on the location of your cpp file but the current working directory when you're running your executable. Let's say you have your executable ready after running the makefile and you [...]
I did put the XML file in the same directory as the executable, but it's still the same error.I succeeded with absolute paths.

closed 2022-10-24 by facontidavide