BehaviorTree/BehaviorTree.CPP #431: Building Behaviour Trees Programatically

PeterQFR, 2022-09-22:
This is a reopening of #276 
In effect I am managing a behaviour tree that comes from some external user, and I want to splice that into a BT that is performing a default behaviour. 

I know I could use a State Machine to do this, but we're using BTs and they should be able to handle it. 

Is there a way to insert a BT defined as an XML string or already processed by the BTFactory, into a running BT?

PeterQFR, 2022-09-22:
Is this use case best handled by modifying the XML directly through a parser or just use a state machine?

facontidavide, 2022-09-23:
First of all, I think you should be familiar with the solution described in the second half of [tutorial 7](https://www.behaviortree.dev/tutorial_07_multiple_xml/).

You do not want to use include in your use case but instead `BehaviorTreeFactory::registerBehaviorTreeFromText`, that can be invoked multiple times.

NOTE that if you call again `registerBehaviorTreeFromText`, the old trees definition with the same name will replaced the old ones. 

now...

 ## Trees are CAN NOT be changed at runtime !!!

BehaviorTrees are code, changing a BehaviorTrees at runtime is like rewriting the lines of a Python script at runtime!

What you **can** do is **destroying completely** the old tree and create a new one.
This means that the state of the single Nodes and the values on the blackboard are reset.

Before you say: "how could it be that I can not change trees at runtime?", remember that a new tree may have a completely different relationship between ports, and figuring out what changed and what persisted is very complex.

I hope this helps. If the solution I am proposing won't work for you, you can reach me out by email to discuss this in more details.
( davide.faconti@gmail.com )

closed 2022-09-23 by facontidavide

PeterQFR, 2022-09-29:
Hi @facontidavide thanks for the clarification. I hadn't noticed the use case used in Tutorial 7, its interesting but again needs all the definitions defined at the start. 

I've gone with a switch between trees and instantiating a new one when it is provided. using `haltTree' to perform some sort of clean up before deletion in an atomic switch arrangement.