BehaviorTree/BehaviorTree.CPP #433: Declared type [int] != current type [void] exception

eschembor-irobot, 2022-09-23:
The full exception that is thrown is
> Blackboard::set() failed: once declared, the type of a port shall not change. Declared type [int] != current type [void]

Here is an example tree that reproduces this issue:

    <BehaviorTree ID="Subtree1">
        <Decorator ID="Repeat" num_cycles="{port_to_use}">
            <Action ID="AlwaysSuccess"/>
        </Decorator>
    </BehaviorTree>
    
    <BehaviorTree ID="Subtree2">
        <Action ID="SetBlackboard" output_key="test_port" value="{port_to_read}"/>
    </BehaviorTree>
    
    <BehaviorTree ID="TestTree">
        <Control ID="Sequence">
            <Action ID="SetBlackboard" output_key="test_port" value="1"/>
            <SubTree ID="Subtree1" port_to_use="test_port"/>
            <SubTree ID="Subtree2" port_to_read="test_port"/>
        </Control ID="Sequence">
    </BehaviorTree>

I think what is happening is that "Subtree1" is first reading from test_port as an int, whereas "Subtree2" is then reading from test_port as a void-typed port. Switching the order so that "Subtree2" is invoked before "Subtree1" does not reproduce the exception.

Should this really throw an exception? Should we not throw this exception if the port isn't changing to a void type?

facontidavide, 2022-09-23:
I was specifically working on this issue today (in the V4 branch).
Once it is fixed, I will backport those changes

facontidavide, 2022-09-23:
Fixed :) 
Thanks

closed 2022-09-23 by facontidavide