BehaviorTree/BehaviorTree.CPP #434: Output port error

laurafbec, 2022-09-25:
Hi everyone
I'm trying to use an output port.
I've initialized it by using

`setOutput("num_poses", num_poses);`

and declare it

`BT::PortsList SetPoses::providedPorts()
{
    return { BT::OutputPort<int>("num_poses") };
}`

but when trying to use it on my BT, 

`<SetPoses name="set_poses" num_poses="{num_poses}"/>`

I get the next error:

`[navigator_node-1]   what():  Possible typo? In the XML, you tried to remap port "num_poses" in node [SetPoses / set_poses], but the manifest of this node does not contain a port with this name.
`

What could I been missing?
Thanks in advance!

facontidavide, 2022-09-25:
I think you forgot **static** in:

```c++
static BT::PortsList SetPoses::providedPorts() { return { BT::OutputPort<int>("num_poses") }; }
```

laurafbec, 2022-09-26:
Thanks @facontidavide!
Actually, I've static in the .h

`class SetPoses : public BT::SyncActionNode
{
  public:
    SetPoses(const std::string& name, const BT::NodeConfiguration& config);
    BT::NodeStatus tick() override;
    static BT::PortsList providedPorts();
};
`
I don't know what I'm missing...Thanks again

facontidavide, 2022-09-26:
send me a project I can compile and I will help

laurafbec, 2022-09-26:
Thanks @facontidavide! I've shared it through google drive

facontidavide, 2022-09-26:
```c++
        void create_behavior_tree() {
            // Build a behavior tree from XML and set it up for logging
            BT::BehaviorTreeFactory factory;
            factory.registerNodeType<GoToPose>("GoToPose");
            factory.registerNodeType<GoToPose>("SetPoses");
            factory.registerNodeType<GoToPose>("SetLocations");
```

Well...

[image]

facontidavide, 2022-09-26:
My suggestion is that you make more active use of interactive debuggers (GDB) to find this kind of issues :smile:

closed 2022-09-26 by facontidavide

laurafbec, 2022-09-26:
:D Thanks Davide!!! I'll do it!