BehaviorTree/BehaviorTree.CPP #436: How to use enum object as input variable in SwitchNode?

valaxkong, 2022-09-28:
I have a SwitchNode like this.
[image]
I need it switching according to an enumerate input variable {var}.
I have following code to cast the enum to int. Finally, i got exception like this:
[image]
How can I do this？Only string is supported as switch variable?

facontidavide, 2022-09-28:
Can you send me code to reproduce this error?

facontidavide, 2022-09-28:
Ideally, a failing unit test ;)

facontidavide, 2022-11-19:
Reopen the issue once you provide an example I can run :)

closed 2022-11-19 by facontidavide