BehaviorTree/BehaviorTree.CPP #445: empty log

laurafbec, 2022-10-15:
Hi everyone,
I'm starting studying and learning this lib. I'm using the next code to log changes on node's status:

`std::string tree_file = "nav_through_poses.xml";
               
            tree_ = factory.createTreeFromFile(bt_xml_dir + "/" + tree_file);

            // Logger to print state changes on console
            BT::StdCoutLogger logger_cout(tree_);

            // Logger to save state changes on file
            BT::FileLogger logger_file(tree_, "bt_trace.fbl");

            // Logger to store execution time of each node
            BT::MinitraceLogger logger_minitrace(tree_, "bt_trace.json");
`

but I'm not getting any info...should I make another call?
Thanks in advance

laurafbec, 2022-10-16:
Solved!

closed 2022-10-16 by laurafbec

salmagro, 2023-11-07:
@laurafbec  Could you please explain how you solve it? I am facing a similar problem.