BehaviorTree/BehaviorTree.CPP #447: parent node

laurafbec, 2022-10-18:
Hi everyone
I would like to know how I can get a given node's parent. I don't find examples of how to 'move' on the tree.
Thanks again

facontidavide, 2022-10-18:
You are not supposed to know who is your parent, or to change behavior based on that.
That would break the encapsulation of the logic

closed 2022-10-18 by facontidavide

laurafbec, 2022-10-19:
Thanks, @facontidavide! The aim was for logging purposes, not to break the encapsulation of the logic.

facontidavide, 2022-10-19:
If you give me a more specific description of the problem and how you want to solve it, I can think about it and give different advices.

laurafbec, 2022-10-19:
Thanks, @facontidavide. I'm starting to play with the code and it seems to me that, to get a better understanding of why a node gets o remains on a certain status, it would be a good idea to get some knowledge about its ancestors. For example, if I want to know why a node has the IDLE status, knowing that his parent is a fallback node that has previously achieved the success status would help to know the paths or alternatives that the execution is following without the need to check Groot. I don't if you see the point of getting this information. There are only ideas :D. Thanks again.

laurafbec, 2022-10-27:
Hi @facontidavide! could you give some advice about this point? Thaks again