BehaviorTree/BehaviorTree.CPP #448: Version 3.8 still working with Groot?

gbcarlos, 2022-10-19:
Hi,

I recently stumbled over the deprecation note of Groot due to the API change of BehaviorTree Cpp. Now, to make things work again, I tried uninstalling my updated BT version and installed 3.8 again, however I am still encountering the same issue. Is 3.8 supposed to still be working with Groot?

facontidavide, 2022-10-19:
Groot depend on a very specific version of BT.CPP, the one specified in the submodule.

I have no bandwidth to maintain this anymore, sorry. Groot 2.0 will be available in less than 3-4 weeks, you should wait for that.

facontidavide, 2022-10-19:
also, what issue are you encountering? You did not specify

gbcarlos, 2022-10-19:
The same issue you already [commented on today](https://github.com/BehaviorTree/Groot/pull/172), I guess there were more people having it.
So I am using it with ROS and installed it via `sudo apt install ros-noetic-behaviortree-cpp-v3`, uninstalling and installing again the BT library did not work out (is it even installing a different version to v4?) 
Not sure if installing via this command also configures the submodule right, also afaik the submodule is only refering to the master branch?