BehaviorTree/BehaviorTree.CPP #450: Cannot compile a ROS package that depends on BehaviorTree.CPP, `lexy/action/parse.hpp: No such file or directory`

galou, 2022-10-19:
I can compile BehaviorTree.CPP with the latest commit (90c2f47df58b) but not a package that depends on BehaviorTree.CPP. I guess that the error is not in my package but I may be wrong.
```console
$ colcon build --packages-select my_package_that_depends_on_behaviortree_cpp
...
install/include/behaviortree_cpp/scripting/any_types.hpp:23:10: fatal error: lexy/action/
parse.hpp: No such file or directory
   23 | #include "lexy/action/parse.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
```
I have ROS Galactic and cmake version 3.16.3.

Thanks for any hint!

galou, 2022-10-19:
Adding
```cmake
INSTALL( DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}/3rdparty/lexy/include/
    DESTINATION ${BEHAVIOR_TREE_INC_DESTINATION}
    FILES_MATCHING PATTERN "*.h*")
```
To BehaviorTree.CPP's `CmakeLists.txt` solves this issue but it doesn't appear to me to be the correct cmake way to do things.

facontidavide, 2022-10-20:
I will fix that, give some time!

closed 2022-10-24 by facontidavide