BehaviorTree/BehaviorTree.CPP #451: Error compiling version 4.0

LuossD, 2022-10-20:
**BehaviorTree.CPP/3rdparty/lexy/include/lexy/dsl/literal.hpp:344:61: error: invalid use of incomplete type ‘struct lexy::_detail::string_literal<1, string_literal<...auto...> >’**

It seems that there is a problem with the compilation of the updated lexy. How can I solve this problem?

facontidavide, 2022-10-20:
Thanks for mentioning, I will try to fix this soon

ShadowWolf9891, 2022-10-20:
I have a similar problem. template parameter "Encoding" is not used in or cannot be deduced from the template argument list of class template "lexy::_detail::lit_trie_matcher<Trie, CurNode>"	Invisible_Girl	
BehaviorTree.CPP-4.0\3rdparty\lexy\include\lexy\dsl\literal.hpp

laurafbec, 2022-10-23:
I managed to compile by installing the last version of lexy lib and updating cmake on Ubuntu 20.04

facontidavide, 2022-10-23:
You should not. I must hide it has private API, not public dependency.

I just came back from Japan let me address this

laurafbec, 2022-10-23:
No problem. I've also had problems with tutorial 3. I get the error 
`terminate called after throwing an instance of 'BT::RuntimeError'
  what():  Error at line 6: -> Node not recognized: SetBlackboard
Aborted (core dumped)`

I don't know if is related to the install process I  made also

facontidavide, 2022-10-24:
Problem should be fixed by https://github.com/BehaviorTree/BehaviorTree.CPP/commit/0cef2e22408a96d80389238cc7bc8514e6da28a7

LuossD, 2022-10-25:
> You should not. I must hide it has private API, not public dependency.
> 
> I just came back from Japan let me address this

I just updated it， but there are still compilation errors : 3rdparty/lexy/include/lexy/dsl/symbol.hpp:63:66: error: invalid use of incomplete type ‘struct lexy::_detail::string_literal<1, string_literal<...auto...> >’
   63 |     return map<_detail::to_type_string<_detail::type_string, SymbolString>>(LEXY_FWD(args)...);

Plenty of similar mistakes with lexy!

facontidavide, 2022-10-25:
Are this error displayed while compiling BT. CPP or when compiling your application that link to my library?

LuossD, 2022-10-25:
> Are this error displayed while compiling BT. CPP or when compiling your application that link to my library?

Just BT. CPP.

facontidavide, 2022-10-25:
This sucks! Can you provide more information about the compiler you are using and the OS?

LuossD, 2022-10-25:
> This sucks! Can you provide more information about the compiler you are using and the OS?

Ubuntu18.04.6 LTS and cmake version 3.17.5

facontidavide, 2022-10-25:
Sorry, but I am not sure that compiler is supported anymore 😢

facontidavide, 2022-10-25:
I confirm that new version requires Ubuntu 20.04 or later. Sorry, the price of progress.
You should continue using **branch v3.8**

closed 2022-10-25 by facontidavide

LuossD, 2022-10-25:
> I confirm that new version requires Ubuntu 20.04 or later. Sorry, the price of progress. You should continue using **branch v3.8**

Okay! Thanks a lot!

laurafbec, 2022-10-27:
> 
sorry @facontidavide! After reinstalling the library over Ubuntu 20.04 without the need to install Lexy lib, I continue with the error
`terminate called after throwing an instance of 'BT::RuntimeError'
  what():  Error at line 6: -> Node not recognized: SetBlackboard
Aborted (core dumped)
`
when testing tutorial 3.
Thanks again