BehaviorTree/BehaviorTree.CPP #452: Failed to register a simple action node when it take a boost::circular_buffer* as input.

valaxkong, 2022-10-23:
I have code like this
``` cpp
factory.registerSimpleAction(
    "XXX",
    [](TreeNode& parent_node) -> NodeStatus {
        ...
        boost::circular_buffer<double>* collision_status_buffer;
        parent_node.getInput("collision_status_buffer_port",
                              collision_status_buffer);
        ...,
        BT::InputPort<boost::circular_buffer<double>*>(
              "collision_status_buffer_port")});
```

Segmentfault arise here.
Is this a known problem or new? Did I use it in a wrong way?

facontidavide, 2022-10-23:
Node with ports must use inheritance, or port are not registered. Also... Which node is actually allocating the circular buffer? Using raw pointer looks dangerous too.

facontidavide, 2022-10-23:
The line

BT::InputPort<boost::circular_buffer<double>*>(
              "collision_status_buffer_port")});

valaxkong, 2022-10-24:
> Node with ports must use inheritance, or port are not registered. Also... Which node is actually allocating the circular buffer? Using raw pointer looks dangerous too.

So, do you mean that the following api is deprecated  ? @facontidavide 
```cpp
void BehaviorTreeFactory::registerSimpleAction(const std::string& ID,
                                               const SimpleActionNode::TickFunctor& tick_functor,
                                               PortsList ports)
```
I just use an initializer list to setup input parameter 'PortsList ports'.
There are some typos in preceding code. Correct proceding code as follows:
```cpp
    factory.registerSimpleCondition(
        "XXX",
        [](TreeNode& parent_node) -> NodeStatus {
            ...
            boost::circular_buffer<double>* collision_status_buffer;
            parent_node.getInput("collision_status_buffer_port",
                                 collision_status_buffer);
            ...
        },
        {...,
         BT::InputPort<boost::circular_buffer<double>*>(
             "collision_status_buffer_port")});
```

facontidavide, 2022-10-24:
ok, I see.

Can you send a simple application that I can compile and run?

closed 2023-02-03 by facontidavide