BehaviorTree/BehaviorTree.CPP #453: Proposal: add library version info in the <root> tag

SubaruArai, 2022-10-25:
With the v4 release including breaking changes to the xml schema, I expect some people will try to run v3 xml on v4 code, or vice versa.
We can add an attribute specifying the targeted library major version into the `<root>` tag to prevent this.
As a bonus, the current implementation of the library doesn't check this filed, so it should be backwards compatible.

e.g.

```xml
<!-- current root tag -->
<root main_tree_to_execute="MainTree" >
    <!-- BTs, node descriptions... -->
</root>

<!-- proposed root tag for v3 -->
<root main_tree_to_execute="MainTree" library_version="3" >
    <!-- BTs, node descriptions... -->
</root>

<!-- proposed root tag for v4 -->
<root main_tree_to_execute="MainTree" library_version="4" >
    <!-- BTs, node descriptions... -->
</root>
```

Also, if anyone has a better idea than `library_version`, feel free to suggest.

@facontidavide are you ok with this idea?

facontidavide, 2022-11-19:
I just did it and changed all the examples.
I still need to support V· files, though

facontidavide, 2023-02-03:
done for version 4. Thanks for the suggestion

closed 2023-02-03 by facontidavide