BehaviorTree/BehaviorTree.CPP #461: CMake is unable to find BehaviorTreeV3 since upgrading to v3.8

koonpeng, 2022-11-02:
The latest release of BehaviorTree in ROS humble changed the package config file from `BehaviorTreeV3Config.cmake` to `behaviortree_cpp_v3Config.cmake`, this causes current projects using BehaviorTree via cmake's `find_package` to break.

facontidavide, 2022-11-02:
sorry about that.

On the other hand... it seems like there is an easy fix, isn't it?

koonpeng, 2022-11-03:
Yeah, it can be easily fixed by looking for `behaviortree_cpp_v3` instead. But given that this is not a major release, I thought this change is not intended.

facontidavide, 2022-11-22:
I can understand you frustration, anyway that name was "wrong" in the first place...

Do you think it would be possible to add BOTH the names simultaneously? I mean adding (not replacing):

```
INSTALL(TARGETS ${BEHAVIOR_TREE_LIBRARY}
    EXPORT BehaviorTreeV3Config
    ARCHIVE DESTINATION ${BEHAVIOR_TREE_LIB_DESTINATION}
    LIBRARY DESTINATION ${BEHAVIOR_TREE_LIB_DESTINATION}
    RUNTIME DESTINATION ${BEHAVIOR_TREE_BIN_DESTINATION}
    )

install(EXPORT BehaviorTreeV3Config
    DESTINATION "${BEHAVIOR_TREE_LIB_DESTINATION}/BehaviorTreeV3/cmake"
    NAMESPACE BT::)

export(TARGETS ${PROJECT_NAME}
    NAMESPACE BT::
    FILE "${CMAKE_CURRENT_BINARY_DIR}/BehaviorTreeV3Config.cmake")
 
```

facontidavide, 2022-11-22:
I think this is the solution to your problem.
I think that some ROS people will hate me, but that's life...

https://github.com/BehaviorTree/BehaviorTree.CPP/commit/4ddc0b035ca70547132ed5b90d9106a53bb46bc7

closed 2022-11-22 by facontidavide

facontidavide, 2022-11-23:
Sorry, but that doesn't work. if anybody with more knowledge about cmake wants to help he/she is very welcome.

I think the best course of action is to keep the new name **behaviortree_cpp_v3**