BehaviorTree/BehaviorTree.CPP #465: Behavior Tree CPP v3 spelling is wrong in latest ROS melodic distribution list. Possible breaking change?

swarajpeppermint, 2022-11-03:
Seems like the spelling of behavior tree cpp v3 is incorrect in the latest melodic distribution list. Is this a breaking change for existing melodic Robotics Stacks? 
[image]

facontidavide, 2022-11-03:
it has been like that since the beginning, and I never changed it

facontidavide, 2022-11-22:
This issue will solve itself May 23rd 2023

http://wiki.ros.org/Distributions

Seriously though, people have been using the misspelled name all this time; if that is an issue, the safest option is to release a new package with the right name alongside the old one.

closed 2022-11-22 by facontidavide