BehaviorTree/BehaviorTree.CPP #466: Suggestion about AsyncSequenceNode and AsyncReactiveSequenceNode

valaxkong, 2022-11-07:
I have been using BehaviorTree.CPP to develop a robotic application for several months.
My behavior tree is reactive and non-reactive mixed. In the reactive part, I frequently meet a behavior pattern that is keepping checking a sequence of async conditions and/or excuting  a sequence of async actions. Specifically, asynchronous nodes means they could return RUNNING during one tick cycle.(Besides, I strongly agree with your renaming about AsyncActionNode as ThreadedAction in v4.0)
So I suggest an AsyncReactiveSequenceNode whose pseudo code is like this:
```cpp
NodeStatus AsyncReactiveSequenceNode::tick()
{
    const size_t children_count = children_nodes_.size();

    setStatus(NodeStatus::RUNNING);

    while (current_child_idx_ < children_count)
    {
        TreeNode* current_child_node = children_nodes_[current_child_idx_];
        const NodeStatus child_status = current_child_node->executeTick();

        switch (child_status)
        {
            case NodeStatus::FAILURE:
            {
                // Reset on failure
                haltChildren();
                current_child_idx_ = 0;
                return child_status;
            }
            case NodeStatus::RUNNING: [[fallthrough]]
            case NodeStatus::SUCCESS:
            {
                current_child_idx_++;
            }
            break;

            case NodeStatus::IDLE:
            {
                throw LogicError("A child node must never return IDLE");
            }
        }   // end switch
    }       // end while loop

    // The entire while loop completed. This means that all the children returned SUCCESS.
    if (current_child_idx_ == children_count)
    {
        haltChildren();
        current_child_idx_ = 0;
    }
    return NodeStatus::SUCCESS;
}

```
This node simply consider child's return RUNNING as SUCCESS. 
AsyncSequenceNode could be defined in a similar way.
I can give a PR if you think my suggestion is constructive.

valaxkong, 2022-11-07:
Oh, sorry! I found ParallelNode can satisfy my requirement.

valaxkong, 2022-11-07:
ParallelNode almost satisfy my requirement, but it has a skiplist mechanism. I think an AsyncSequenceNode mentioned above is still needed.

facontidavide, 2023-10-16:
Assuming his error is obsolete/stale. Closing

closed 2023-10-16 by facontidavide