BehaviorTree/BehaviorTree.CPP #467: BT Runtime SegmentFault

BeyondCompare, 2022-11-08:
the tree defination as follows:
<root main_tree_to_execute = "MainTree" >                                                                          
             <BehaviorTree ID="MainTree">                                                                                   
               <Sequence name="root">                                                                                      
                   <PrintTarget     target="{GoalPosition}"/>                                                               
                    <SetBlackboard   output_key="OtherGoal" value="-1;3" />                                                 
                    <PrintTarget     target="{OtherGoal}" />                                                                
                </Sequence>                                                                                                 
             </BehaviorTree>                                                                                                
         </root>                                                                                                            
         )";           

and then i build the tree just like this:

     BT::BehaviorTreeFactory factory;                                                                                    
           factory.registerNodeType<TestTreeNode>("PrintTarget");                                                            
           auto tree = factory.createTreeFromText(xmlFile); 

but when i run the code, it segmentfault

[image]

[image]

it's confused

BeyondCompare, 2022-11-08:
the tree just like this:
static const char* xml_text = R"(                                                                                   
                                                                                                                            
         <root main_tree_to_execute = "MainTree" >                                                                          
             <BehaviorTree ID="MainTree">                                                                                   
                <Sequence name="root">                                                                                      
                   <PrintTarget     target="{GoalPosition}"/>                                                               
                    <SetBlackboard   output_key="OtherGoal" value="-1;3" />                                                 
                    <PrintTarget     target="{OtherGoal}" />                                                                
                </Sequence>                                                                                                 
             </BehaviorTree>                                                                                                
         </root>                                                                                                            
         )";

facontidavide, 2022-11-08:
`SetBlackboard` was removed from 4.0 (that maybe was not a great idea)
Which version are you using?
Can you share a code that I can compile and test?

BeyondCompare, 2022-11-08:
> 

> `SetBlackboard` was removed from 4.0 (that maybe was not a great idea) Which version are you using? Can you share a code that I can compile and test?
> 
> If you share a simple project, I can fix this today

I'm sorry ， it's my fault,  i forget to install bt in ros envioment,  i had resolve this compile problem, thank you handsome facontidavide